wb2osz/direwolf #131: direwolf 100% CPU occupancy with -n 1 option

f6bvp, 2018-01-27:
Context is Raspberry Pi 3 (quadcore) direwolf dev 1.5, Compass Linux 4.9.35-v7+ with UDRC II modem HAT board.
direwolf.conf options are standard for UDRC with ACHANNELS 2 as described in NW digital Radio doc.
Both CHANNELS speed are set to 1200 bauds.
Starting direwolf with simplest command line -t 0 and monitoring system with top utility gives less than 19% CPU occupancy.
With one CHANNEL speed is set to 9600 then CPU % raises up to 75% on average with rare peaks to 99% when a packet APRS frame is received and decoded.
With both CHANNEL set at 9600 then %CPU is 99.8-100.2 % and error message is displayed : 
"Audio input device 0 error: Broken pipe" certainly related to CPU overload.
However changing direwolf.conf ACHANNELS 1 gives also constant 100 % CPU occupancy !
The same constant 100% CPU is obtained if entering option -n 1 with ACHANNELS 2 in direwolf.conf
even if both CHANNEL(s) are set to 1200 bauds.
Conclusion : a "wait" instruction may be missing somewhere in the code to let soft interrupts occur when single audio channel is selected.
Regards, 
Bernard

T-Shilov, 2022-09-14:
1. Raspberry PI 3B v1.2
2. direwolf stable,now 1.6+dfsg-1+b1 arm64 
3. 2022-09-06-raspios-bullseye-arm64-lite.img.xz

One core is also constantly consuming 100%.
Why won't anyone fix this problem?

dranch, 2022-09-15:
@T-Shilov Is your issue that you configure two channels in the direwolf.conf file get you override it to only use one channel by using the "-n 1" option when starting Direwolf?  If correct, is there a reason why you just don't configure direwolf.conf to meet your specific needs vs. override it's settings at the command line?

wb2osz, 2025-03-30:
This is from 7 years ago.
Efficiency improvements have been made in 9600 decoding.
Open a new case if this is a still a problem.

closed 2025-03-30 by wb2osz