wb2osz/direwolf #134: dw-start.sh not working in Raspbian Stretch

na7q, 2018-02-25:
Looks like dw-start.sh is not working in Raspbian Stretch. Terminal opens and is labeled "Direwolf", but the window is black and has not actually started the application. Doesn't work from frontal or manually running the script.

dranch, 2018-03-03:
You are probably missing a required package.  If you run it manually, do you see an error?

--David
KI6ZHD

na7q, 2018-03-04:
Don't know what package I'd be missing. It runs manually just fine without the script. 
Using the script with crontab or rc.local or anything method, it doesn't load Direwolf in any way. 
It strangely produces no errors when looking at the logs. 
I ended up just running it directly with rc.local for my usage.

dranch, 2018-03-04:
If you try again using it from rc.local, can you send me the output of the /var/tmp/dw-start.log file when it fails?

na7q, 2018-03-05:
It results in this 100% of the time, repeating forever.

Mon Mar  5 00:29:05 UTC 2018
Direwolf in GUI mode start up
DISPLAY=:0
-----------------------
Mon Mar  5 00:29:58 UTC 2018
Direwolf in GUI mode start up
DISPLAY=:0
-----------------------
Mon Mar  5 00:39:05 UTC 2018
Direwolf in GUI mode start up
DISPLAY=:0
-----------------------
Mon Mar  5 00:46:01 UTC 2018
Direwolf in GUI mode start up
DISPLAY=:0
-----------------------

dranch, 2018-03-05:
Look in the script itself and tell me what version you're using?  Is it v1.3?  There should be more there.  If you run the command from within a terminal manually, what do you see?

na7q, 2018-03-05:
I'm using v1.4. When I run the command "direwolf", it loads perfectly as it should. When I run the script dw-start.sh I get a window labeled Dire Wolf, but it's black, no text, a flashing box, and direwolf not actually running. The terminal window stays open but never does anything. I can run direwolf in another terminal while the one is open, telling me it's not actually running.

dranch, 2018-03-05:
Ok.. please upgrade to v1.4 as there was a LOT of fixes.. even better, try out the new v1.5 Beta.  Next, the window opening is what is expected. in black but can you try to figure out what the flashing text says?

na7q, 2018-03-05:
I thought you meant the version of the dw-start.sh? 
I'm on v1.4 and v1.5 Beta of Direwolf on two Pi's with stretch. Even the updated v1.5 Beta still has the same issue I'm describing.

dranch, 2018-03-05:
Ahh.. sorry, my mistake.  Anyway, any chance of decoding the flashing text?  Sounds like Direwolf is starting up though.

na7q, 2018-03-05:
Honestly, I haven't got a clue as to the flashing box. It just acts like it's going to start. So the script is running to a point. I'll have to figure it out another day when I get some time to sit down to it.
So for now, running it in rc.local works the way I need. I'm up and running direwolf and connected over bluetooth to my phone in 16 seconds flat. I can't complain too much.

G0UWS, 2018-09-07:
Did this a fix for this become known.??

I have just built my Pi with Stretch and Direwolf 1.4 and am finding the exact same issue.

Thought I would ask before I rebuild it.

dranch, 2018-09-07:
Which version of Raspbian Stretch are you running?  The full version with Xwindows or the lite version with no UI?  Does Direwolf run manually without using the script?  Please try using Direwolf 1.5Beta4 and see if you're still seeing the same issue

G0UWS, 2018-09-07:
I am using the latest GUI based version updated yesterday.

Direwolf can manually is working fine, Like the original post when ran via Cron the Direwolf terminal window opens but no text is displayed at all. Not even the initial Direwolf bit with igate communication established confirmations.

I will upgrade to 1.5 beta 4 and confirm.

na7q, 2018-09-09:
I never got it working. So I just run it via rc.local. No window to view it in, but I at least get it auto started as needed.

MacroPowerPC, 2018-10-26:
I got same issue when using direwolf 1.5 on Stretch.  It seems dw-start.sh open a lxterminal windows but execute nothing.   it could be fixed by add option "-l" after lxterminal :

/usr/bin/lxterminal -l -t "Dire Wolf" -e "$DWCMD" & 

:)

MacroPowerPC, 2018-10-27:
it could be started normally by add option "-l" to command lxterminal in dw-start.sh :
...
   if [ -x /usr/bin/lxterminal ]; then
      /usr/bin/lxterminal **-l** -t "Dire Wolf" -e "$DWCMD" &
      SUCCESS=1
...

khkremer, 2019-01-09:
I am having the same problem when using the CLI version: Direwolf does not start up (or, it does not keep on running). As the others have mentioned, I can run it without a problem from the command line. I am using rtl_fm  to pipe it's output into direwolf. This is with direwolf version 1.5 - just updated from git. 

This is what I am getting when I try to start it:

pi@APRSGW:~ $ ./dw-start.sh 
Direwolf in CLI mode start up
There is a screen on:
	1945.direwolf	(01/09/2019 10:47:10 AM)	(Detached)
1 Socket in /run/screen/S-pi.
-----------------------
pi@APRSGW:~ $ !ps
ps auxww | grep dire
pi        1952  0.0  0.0   4376   580 pts/0    S+   10:47   0:00 grep --color=auto dire
pi@APRSGW:~ $ screen -ls
No Sockets found in /run/screen/S-pi.

The last two commands are to see if direwolf is actually running, or if the screen session is still alive. 

This is with Raspbian Stretch, also updated today.

MisterBearBear, 2019-03-03:
I found the solution. Dire Wolf can now start automatically in GUI mode under Raspbian Stretch. You need the following changes in the dw-start.sh:

https://github.com/wb2osz/direwolf/blob/master/dw-start.sh

Line 38: DIREWOLF="/usr/local/bin/direwolf"

Line 46: DWCMD="bash -l -c '$DIREWOLF -c /home/pi/direwolf.conf'"

Line 124: /usr/bin/lxterminal -l -t "Dire Wolf" -e "$DWCMD" &

Hope this helps. At least this works for me. If anyone has better solution, please feel free to share.

73 de VR2ZDX

dranch, 2019-03-04:
I've done a pull request to WB2OSZ that integrates some of these fixes as well as some others.  Let me know if this new script doesn't work for you.  https://github.com/wb2osz/direwolf/pull/199

jamflying, 2019-03-16:
I tried the new dw-start.sh, there is nothing run also...
I run the command in crontab: /home/pi/dw-start.sh
and I got the result:
========= start =========
pi@raspberrypi:~ $ /home/pi/dw-start.sh

dw-start.sh
-----------
Sat Mar 16 21:09:10 CST 2019
Sleeping for 30 seconds to let any boot/reboot delays conclude
Direwolf in CLI mode start up.  All log output recorded to /var/tmp/dw-start.log
 
There is a screen on:
460.direwolf (03/16/2019 09:09:39 PM) (Detached)
1 Socket in /run/screen/S-pi.

You can re-attach to the Direwolf screen with:
     screen -dr direwolf
-----------------------
pi@raspberrypi:~ $ screen -dr direwolf
There is no screen to be detached matching direwolf.

========= end =========

jamflying, 2019-03-16:
I got a solution at issue #119: 
from:
```
$SCREEN -d -m -S direwolf $DWCMD >> $LOGFILE
```
to
```
$SCREEN -d -m -S direwolf bash -c "$DWCMD >> $LOGFILE"
```
Now it works on my [igate BG6LH-10](https://aprs.fi/info/a/BG6LH-10)

dranch, 2019-03-16:
Thanks for the check and I now see the issue I was seeing.  The issue resolved around the fact that  this script supports three ways of start-up.  Manually from the CLI command prompt, manually from an Xwindows terminal, or from cron.   I've improved each mode's start up, it's method of logging, etc.  Please give the new script a try:

   https://github.com/wb2osz/direwolf/pull/199

--David

behem0t, 2019-05-19:
With this new file version I get an error that script can't find direwolf.conf in "/home/pi/" but I provided valid path (in my case "-c /home/pi/direwolf/direwolf.conf") I've edited this path in uncommented line to start with rtl_fm

netsound, 2019-08-18:
> I got a solution at issue #119:
> from:
> 
> ```
> $SCREEN -d -m -S direwolf $DWCMD >> $LOGFILE
> ```
> 
> to
> 
> ```
> $SCREEN -d -m -S direwolf bash -c "$DWCMD >> $LOGFILE"
> ```
> 
> Now it works on my [igate BG6LH-10](https://aprs.fi/info/a/BG6LH-10)

This just helped me get my dw-start.sh script going thanks..

farm2440, 2020-01-12:
Thanks to VR2ZDX! That worked for me too. With one more thing. 
Seems like direwolf can't find libhamlib when starting from script. I was unable to start direwolf from cli as root. 
I ran the command  _$sudo ldconfig_  too to fix it
LZ2SMX, 73

rgrobinson131, 2020-05-26:
I'm trying to build a pi baofeng usb sound card aprs digi following the guide online. I cannot get the auto boot to work for direwolf? Can somebody help please

dranch, 2020-05-26:
Since you've replied to this ongoing Git issue and I assume you've followed the recommendations in this discussion, what is your issue?  Do you have specific errors that you can share with us?