wb2osz/direwolf #136: ioctl HIDIOCGRAWINFO error using CM108 PTT

va7acq, 2018-02-26:
I've been using the DireWolf 1.5 Beta for the CM108 PTT feature (Thank you!), but it shows an error in the logs even though it seems to work and triggers my PTT just fine. The error message may be a false positive.

The exact error message is "ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.", and it repeats multiple times during a session. I'm using RMS Express on Windows 10, and am connecting to DireWolf on my Raspberry Pi 3 over TCP KISS. I'm using the latest build of Raspbian Stretch Lite with all updates applied. The error only appears when I transmit. I've tried running DireWolf as "root", and as the "pi" user, but it makes no difference.

When I first used the CM108 adapter for PTT I simply used `PTT CM108` in my config file as described in the user guide, but DireWolf gave this error in the logs:

`Config file line 18: Could not determine USB Audio GPIO PTT device for audio output plughw:1,0.
You must explicitly mention a device name such as /dev/hidraw1.`

So I changed the config line to `PTT CM08 3 /dev/hidraw0` and now it works, but gives the error message.

Here's some typical log output:
`Feb 26 07:19:13 testpi systemd[1]: Started Dire Wolf TNC.
Feb 26 07:19:13 testpi direwolf[5598]: Dire Wolf version 1.5 (Feb 12 2018) Beta Test 2
Feb 26 07:19:13 testpi direwolf[5598]: Includes optional support for:  gpsd hamlib cm108-ptt
Feb 26 07:19:13 testpi direwolf[5598]: Reading config file /usr/local/etc/direwolf/kiss_1200.conf
Feb 26 07:19:13 testpi direwolf[5598]: Audio device for both receive and transmit: plughw:1,0  (channel 0)
Feb 26 07:19:13 testpi direwolf[5598]: Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 48000 sample rate / 3.
Feb 26 07:19:13 testpi direwolf[5598]: Using /dev/hidraw0 GPIO 3 for channel 0 PTT control.
Feb 26 07:19:13 testpi direwolf[5598]: Ready to accept AGW client application 0 on port 8000 ...
Feb 26 07:19:13 testpi direwolf[5598]: Ready to accept KISS TCP client application 0 on port 8001 ...
Feb 26 07:19:13 testpi direwolf[5598]: Log file is "/home/pi/aprs.log"

Feb 26 07:37:38 testpi direwolf[5598]: Attached to KISS TCP client application 1...
Feb 26 07:37:38 testpi direwolf[5598]: Ready to accept KISS TCP client application 0 on port 8001 ...
Feb 26 07:37:38 testpi direwolf[5598]: KISS protocol set Persistence = 160, port 0
Feb 26 07:37:38 testpi direwolf[5598]: KISS protocol set SlotTime = 30 (*10mS units = 300 mS), port 0
Feb 26 07:37:38 testpi direwolf[5598]: KISS protocol set TXDELAY = 20 (*10mS units = 200 mS), port 0
Feb 26 07:37:38 testpi direwolf[5598]: KISS protocol set FullDuplex = 0, port 0
Feb 26 07:37:46 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:46 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(SABM cmd, p=1)
Feb 26 07:37:46 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:47 testpi direwolf[5598]: VE7SEU audio level = 58(19/16)   [NONE]   ||||||||_
Feb 26 07:37:47 testpi direwolf[5598]: [0.3] VE7SEU>VA7ACQ:(UA res, f=1)
Feb 26 07:37:47 testpi direwolf[5598]: VE7SEU audio level = 38(17/15)   [NONE]   ||||||||_
Feb 26 07:37:47 testpi direwolf[5598]: [0.3] VE7SEU>VA7ACQ:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0);FW: VE7SEU<0x0d>
Feb 26 07:37:48 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:48 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(RR res, n(r)=1, f=0)
Feb 26 07:37:49 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:50 testpi direwolf[5598]: VE7SEU audio level = 35(15/16)   [NONE]   _|||||||_
Feb 26 07:37:50 testpi direwolf[5598]: [0.4] VE7SEU>VA7ACQ:(I cmd, n(s)=1, n(r)=0, p=0, pid=0xf0)[RMS Express-1.5.11.0-B2FHM$]<0x0d>
Feb 26 07:37:51 testpi direwolf[5598]: VE7SEU audio level = 24(14/14)   [NONE]   ||||||||_
Feb 26 07:37:51 testpi direwolf[5598]: [0.3] VE7SEU>VA7ACQ:(I cmd, n(s)=2, n(r)=0, p=0, pid=0xf0)VA7ACQ DE VE7SEU ><0x0d>
Feb 26 07:37:51 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:51 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(I cmd, n(s)=0, n(r)=3, p=0, pid=0xf0);FW: VA7ACQ<0x0d>
Feb 26 07:37:51 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(I cmd, n(s)=1, n(r)=3, p=0, pid=0xf0)[RMS Express-1.5.11.0-B2FHM$]<0x0d>
Feb 26 07:37:52 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:53 testpi direwolf[5598]: VE7SEU audio level = 56(20/15)   [NONE]   ||||||___
Feb 26 07:37:53 testpi direwolf[5598]: [0.2] VE7SEU>VA7ACQ:(RR res, n(r)=2, f=1)
Feb 26 07:37:53 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:53 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(I cmd, n(s)=2, n(r)=3, p=0, pid=0xf0); VE7SEU DE VA7ACQ (CN88HL)<0x0d>
Feb 26 07:37:53 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(I cmd, n(s)=3, n(r)=3, p=0, pid=0xf0)FF<0x0d>
Feb 26 07:37:55 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:56 testpi direwolf[5598]: VE7SEU audio level = 51(20/16)   [NONE]   ||||||___
Feb 26 07:37:56 testpi direwolf[5598]: [0.2] VE7SEU>VA7ACQ:(RR res, n(r)=4, f=1)
Feb 26 07:37:56 testpi direwolf[5598]: VE7SEU audio level = 38(19/15)   [NONE]   |||||||__
Feb 26 07:37:56 testpi direwolf[5598]: [0.3] VE7SEU>VA7ACQ:(I cmd, n(s)=3, n(r)=4, p=0, pid=0xf0)FQ<0x0d>
Feb 26 07:37:57 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:37:57 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(RR res, n(r)=4, f=0)
Feb 26 07:37:57 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:38:12 testpi direwolf[5598]: VE7SEU audio level = 54(19/16)   [NONE]   ||||||___
Feb 26 07:38:12 testpi direwolf[5598]: [0.2] VE7SEU>VA7ACQ:(DISC cmd, p=1)
Feb 26 07:38:13 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Feb 26 07:38:13 testpi direwolf[5598]: [0L] VA7ACQ>VE7SEU:(UA res, f=1)
Feb 26 07:38:13 testpi direwolf[5598]: ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.`

Here's my config file:
`pi@testpi:~ $ cat /usr/local/etc/direwolf/kiss_1200.conf 
ADEVICE plughw:1,0
ACHANNELS 1
ARATE 48000
CHANNEL 0
MODEM 1200
DWAIT 0
SLOTTIME 10
PERSIST 63
TXDELAY 30
TXTAIL 10
FULLDUP OFF
AGWPORT 8000
KISSPORT 8001
DCD GPIO 17
PTT CM108 3 /dev/hidraw0`

Output from your cm108 application:
`pi@testpi:~ $ cm108 
    VID  PID   Product                          Sound                  ADEVICE         HID [ptt]        
    ---  ---   -------                          -----                  -------         ---------        
    0d8c 013c  USB PnP Sound Device             /dev/snd/pcmC1D0c      plughw:1,0      /dev/hidraw0     
    0d8c 013c  USB PnP Sound Device             /dev/snd/pcmC1D0p      plughw:1,0      /dev/hidraw0     
    0d8c 013c  USB PnP Sound Device             /dev/snd/controlC1                     /dev/hidraw0`

lsusb:
`pi@testpi:~ $ lsusb
Bus 001 Device 005: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller`

uname:
`pi@testpi:~ $ uname -a
Linux testpi 4.14.21-v7+ #1095 SMP Fri Feb 23 18:21:29 GMT 2018 armv7l GNU/Linux`

dmesg:
`[    3.091422] usb 1-1.3: new full-speed USB device number 4 using dwc_otg
[    3.230322] usb 1-1.3: New USB device found, idVendor=0d8c, idProduct=013c
[    3.239126] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.248367] usb 1-1.3: Product: USB PnP Sound Device
[    3.255064] usb 1-1.3: Manufacturer: C-Media Electronics Inc.      
[    3.267627] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.3/0003:0D8C:013C.0001/input/input0
[    3.351636] hid-generic 0003:0D8C:013C.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-3f980000.usb-1.3/input3`

This is the USB CM108 audio adapter I am using:
https://www.sanwulasers.org/product/hf07

Let me know if there's any other details I can provide.

73,
VA7ACQ

dm-o, 2018-09-28:
Hey guys,

I have same issue here - after every PTT I get:
"ioctl HIDIOCGRAWINFO failed for /dev/hidraw1. errno = 0."

73, DO6DAD
Steffan

pfiliberti, 2019-05-21:
I had the same issue. The ID of my device was not included in the test to recognize the device in cm108.c.
I added a new define for my ID under #define CMEDIA_PID_CM108AH with the PID of my usb device. Then I modified the if test to detect the device. Look for #define GOOD_DEVICE and you will see the code. The easy fix is to just change the pid in the define for CMEDIA_PID_CM108AH to your id as shown with lsusb and recompile. Hope this helps.

plett, 2019-09-16:
This is fixed by #168

m1geo, 2020-06-02:
For anyone else coming across this issue, I found that changing the permissions of hidraw0 stopped the errors, despite the PTT functioning.

`$ sudo chmod 777 /dev/hidraw0`

George M1GEO

yjf007, 2024-01-14:
My cm108 sound card won't fire beacons，Notice：failed for /dev/hidraw0. errno = 0."

wb2osz, 2024-01-14:
yjf007:

Please provide the following information to help troubleshoot the issue:

(1) operating system.
(2) result of "id" command.
(3) result of "ls -l /dev/hidraw0" command.
(4) result of "cm108" command.
(5) content of your direwolf config file.

73,
John WB2OSZ

yjf007, 2024-01-15:
to：John WB2OSZ
Thank you very much for your reply!
（1）Reading config file /root/direwolf.conf
2024-01-15 10:50:00 Audio input device for receive: stdin  (channel 0)
2024-01-15 10:50:00 Audio out device for transmit: plughw:1,0  (channel 0)
2024-01-15 10:50:00 Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, A+, 24000 sample rate.
2024-01-15 10:50:00 Using /dev/hidraw1 GPIO 3 for channel 0 PTT control.
2024-01-15 10:50:00 Ready to accept AGW client application 0 on port 8000 ...
2024-01-15 10:50:00 Ready to accept KISS TCP client application 0 on port 8001 ...
2024-01-15 10:50:05 Now connected to IGate server china.aprs2.net (121.42.207.32)
2024-01-15 10:50:05 Check server status here http://121.42.207.32:14501
2024-01-15 10:50:07 Attached to KISS TCP client application 0 on port 8001 ...
2024-01-15 10:50:07 Ready to accept KISS TCP client application 1 on port 8001 ...
（2）
2024-01-15 10:50:09 [ig] # aprsc 2.1.15-gc67551b
2024-01-15 10:50:09 [ig] # logresp BG9ITI-10 verified, server T2CS
2024-01-15 10:51:00 [ig] BG9ITI-10>APDW17:!3607.71N/10343.32Er144.640MHz 
2024-01-15 10:51:00 Could not open /dev/hidraw1 for write, errno=2
2024-01-15 10:51:00 ERROR:  PTT for channel 0 has failed.  See User Guide for troubleshooting tips.
2024-01-15 10:51:00 [0L] BG9ITI-10>APDW17:!3607.71N/10343.32Er144.640MHz Hello,test
（3）root@BG9ITI:~# ls -l /dev/hidraw0
ls: cannot access '/dev/hidraw0': No such file or directory
（4）root@BG9ITI:~# cm108
    VID  PID   Product                          Sound                  ADEVICE        ADEVICE            HID [ptt]
    ---  ---   -------                          -----                  -------        -------            ---------
**  0d8c 0012  USB Audio Device                 /dev/snd/pcmC1D0c      plughw:1,0     plughw:Device,0
**  0d8c 0012  USB Audio Device                 /dev/snd/pcmC1D0p      plughw:1,0     plughw:Device,0
**  0d8c 0012  USB Audio Device                 /dev/snd/controlC1

** = Can use Audio Adapter GPIO for PTT.
（5）ADEVICE plughw:1,0
PTT CM108 3 /dev/hidraw1
MYCALL BG9ITI-10
IGSERVER china.aprs2.net
IGLOGIN BG9ITI-10 12345
PBEACON sendto=IG delay=1 every=10 symbol=/r lat=36^07.71N long=103^43.32E freq="144.640" comment="xxx"
PBEACON sendto=0 delay=1 every=10 symbol=/r lat=36^07.71N long=103^43.32E freq="144.640" comment="Hello,test" 
MODEM 1200
AGWPORT 8000
KISSPORT 8001

wb2osz, 2025-03-30:
Looks like a permission problem.
A few years back, a new file was added to set the permissions of the /dev/hdraw part of the CMedia USB audio adapters.
/etc/udev/rules.d/99-direwolf-cmedia.rules
Considering it fixed.  Open a new case if still an issue.

closed 2025-03-30 by wb2osz