wb2osz/direwolf #141: Doesn't indicate audio device in use, no way to verify setting

madsci77, 2018-04-04:
With 1.3.1 under Windows 8.1, no audio input or output is shown as selected.  There's no way to see which input and output are in use.

[image]

dranch, 2018-04-05:
Please post your direwolf.conf which shows which soundcard you've configured per the Direwolf UserGuide.pdf.

--David

madsci77, 2018-04-05:
To clarify, the problem isn't that I can't select a soundcard - it's that it seemed to be working and decoding (at least last time I had it fired up) without any soundcard being displayed as active.

dranch, 2018-04-05:
So you're asking for a feature enhancement to show what soundcard is being used upon start in Windows?   It does show this in Linux:
--
Dire Wolf DEVELOPMENT version 1.5 C (Dec 16 2017)
Includes optional support for:  gpsd hamlib

Reading config file /etc/ax25/direwolf-d710-passall.conf
Line 259: Using a FIX_BITS value greater than 1 is not recommended for normal operation.
Line 259: There is an old saying, "Be careful what you ask for because you might get it."
The PASSALL option means allow all frames even when they are invalid.
You are asking to receive random trash and you WILL get your wish.
Don't complain when you see all sorts of random garbage.  That's what you asked for.
**Audio device for both receive and transmit: plughw:3,0  (channel 0)**
--

madsci77, 2018-04-05:
I'm just saying that it seems like a bug, to have it say that * indicates the active soundcard but a soundcard is active without being indicated.

As minor bugs go, the screen color issue would be a much higher priority for me.  Bright green on white is almost impossible to read.  You can set different colors but it doesn't stay.

dranch, 2018-04-05:
Fair point on the WIndows side of showing what soundcard is being used (at least show what's been configured).  If you prefer different colors, you'll need to make some minor source code changes and recompile.

CInsights, 2018-04-05:
If you prefer no color (which I do) just set black and white with the -t0 command line option.

 

Bob (VK2GJ)

madsci77, 2018-04-05:
There's already a color selection menu in the preferences - it just doesn't stay set and I don't think the background ever changes.  I'll try black and white.

dranch, 2018-04-05:
Please note that Github and this issue tracker is for bugs and maybe enhancement requests, it's not for general support.  To get general support, please join direwolf_packet@yahoogroups.com .

Btw, Direwolf is a command line program that's ONLY configured by direwolf.conf or the command line options you give it.  I have a feeling you're confusing the Windows's Command Prompt window's preferences which can change fonts, colors, etc. with Direwolf.  Those are NOT Direwolf settings.