wb2osz/direwolf #148: allow FIFO file input

Wojtus11, 2018-05-21:
Hi,
It would be great if additional source "devices" could be added, namely arbitrary character devices from filesystem (intended use: FIFO named pipes).

Right now stdin is available, but cannot be used for than once. The other choice is localhost UDP stream, which is very, very unreliable for high data throughput on Raspberry with SDR.

TCP/IP would be great too, but I think reading named FIFO pipe is easier to implement.

Thanks,
Wojtus

isguarded, 2021-05-25:
you could tail -f your pipe to stdin. there is no reason you couldn't use your fifo pipe on direwolf's stdin.

sp9wpn, 2021-05-25:
I am aware of that and indicate the limitation: "_Right now stdin is available, but cannot be used for than once._"