wb2osz/direwolf #149: Feature Request: Remote Programming

na7q, 2018-05-23:
Direwolf would be a great replacement as a digipeater for use in remote sites if it could accept remote programming using messaging. Something like the KPC units, or Alinco TNC units, remote programming would be great for Direwolf to have. Making it a better option for remote sites without internet access.

wb2osz, 2019-11-24:
Could you give a more specific example of what you want to do?
Do you want to change the configuration over the radio?

na7q, 2019-12-01:
Correct, change the configuration of direwolf over the radio similar to that of Argent Data TNCs. For us that use direwolf in very remote access locations, this would be amazing to have implemented.
https://www.argentdata.com/catalog/product_info.php?products_id=166