wb2osz/direwolf #150: Compile error in Alpine Linux 3.7

sgub, 2018-05-23:
OS: Alpine Linux 3.7
gcc-6.4.0-r5
linux-headers-4.4.6-r2
alsa-lib-dev-1.1.4.1-r2
...
direwolf-1.5-beta2

Same error on both x86_64 and armv6l (raspberry pi)

```
/tmp/direwolf-1.5-beta2 # make
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA   -c -o direwolf.o direwolf.c
/usr/include/string.h:83:45: error: expected declaration specifiers or '...' before string constant
 size_t strlcat (char *, const char *, size_t);
                                             ^
In file included from direwolf.c:41:0:
direwolf.h:294:65: error: expected declaration specifiers or '...' before '__func__'
 #define strlcat(dst,src,siz) strlcat_debug(dst,src,siz,__FILE__,__func__,__LINE__)
                                                                 ^
/usr/include/string.h:83:45: error: expected declaration specifiers or '...' before numeric constant
 size_t strlcat (char *, const char *, size_t);
                                             ^
/usr/include/string.h:84:45: error: expected declaration specifiers or '...' before string constant
 size_t strlcpy (char *, const char *, size_t);
                                             ^
direwolf.h:293:65: error: expected declaration specifiers or '...' before '__func__'
 #define strlcpy(dst,src,siz) strlcpy_debug(dst,src,siz,__FILE__,__func__,__LINE__)
                                                                 ^
/usr/include/string.h:84:45: error: expected declaration specifiers or '...' before numeric constant
 size_t strlcpy (char *, const char *, size_t);
                                             ^
In file included from /usr/include/fortify/stdio.h:23:0,
                 from direwolf.c:45:
/usr/include/fortify/string.h:156:1: error: 'strlcat' undeclared here (not in a function)
 _FORTIFY_FN(strlcat) size_t strlcat(char *__d, const char *__s, size_t __n)
 ^
In file included from direwolf.c:50:0:
/usr/include/fortify/string.h: In function 'strlcat':
/usr/include/fortify/string.h:162:9: error: called object '__orig_strlcat' is not a function or function pointer
  return __orig_strlcat(__d, __s, __n);
         ^~~~~~~~~~~~~~
In file included from /usr/include/fortify/stdio.h:23:0,
                 from direwolf.c:45:
/usr/include/fortify/string.h:156:1: note: declared here
 _FORTIFY_FN(strlcat) size_t strlcat(char *__d, const char *__s, size_t __n)
 ^
/usr/include/fortify/string.h: At top level:
/usr/include/fortify/string.h:165:1: error: 'strlcpy' undeclared here (not in a function)
 _FORTIFY_FN(strlcpy) size_t strlcpy(char *__d, const char *__s, size_t __n)
 ^
In file included from direwolf.c:50:0:
/usr/include/fortify/string.h: In function 'strlcpy':
/usr/include/fortify/string.h:171:9: error: called object '__orig_strlcpy' is not a function or function pointer
  return __orig_strlcpy(__d, __s, __n);
         ^~~~~~~~~~~~~~
In file included from /usr/include/fortify/stdio.h:23:0,
                 from direwolf.c:45:
/usr/include/fortify/string.h:165:1: note: declared here
 _FORTIFY_FN(strlcpy) size_t strlcpy(char *__d, const char *__s, size_t __n)
 ^
make: *** [<builtin>: direwolf.o] Error 1
```

dranch, 2018-05-23:
No problem compiling 1.5 Beta2 on Raspbian Stretch which has GCC 6.3.0-18+rpi1+deb9u1 with the following support:
        >       This includes support for gpsd.
        >       This does NOT include support for hamlib.
        >       This does NOT include support for CM108/CM119 PTT.

Could you try again with the following commands:

   make clean
   make -f Makefile.linux tocalls-symbols
   make -f Makefile.linux -j4

--David
KI6ZHD

sgub, 2018-05-24:
I tried the commands, and got more errors...

```
/tmp/direwolf-1.5-beta2 # apk add alpine-sdk linux-headers alsa-lib-dev diffutils
...
/tmp/direwolf-1.5-beta2 # make clean
...
/tmp/direwolf-1.5-beta2 # make -f Makefile.linux tocalls-symbols
...
/tmp/direwolf-1.5-beta2 # make -f Makefile.linux -j4
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA   -c -o direwolf.o direwolf.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA   -c -o config.o config.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA   -c -o recv.o recv.c
echo " " > tune.h
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DGEN_FFF -o gen_fff demod_afsk.c dsp.c textcolor.c -lm -lpthread -lrt -lasound
/usr/include/string.h:83:45: error: expected declaration specifiers or '...' before string constant
 size_t strlcat (char *, const char *, size_t);
                                             ^
In file included from config.c:37:0:
direwolf.h:294:65: error: expected declaration specifiers or '...' before '__func__'
 #define strlcat(dst,src,siz) strlcat_debug(dst,src,siz,__FILE__,__func__,__LINE__)
                                                                 ^
/usr/include/string.h:83:45: error: expected declaration specifiers or '...' before numeric constant
 size_t strlcat (char *, const char *, size_t);
                                             ^
/usr/include/string.h:84:45: error: expected declaration specifiers or '...' before string constant
 size_t strlcpy (char *, const char *, size_t);
                                             ^
direwolf.h:293:65: error: expected declaration specifiers or '...' before '__func__'
 #define strlcpy(dst,src,siz) strlcpy_debug(dst,src,siz,__FILE__,__func__,__LINE__)
                                                                 ^
/usr/include/string.h:84:45: error: expected declaration specifiers or '...' before numeric constant
 size_t strlcpy (char *, const char *, size_t);
                                             ^
/usr/include/string.h:83:45: error: expected declaration specifiers or '...' before string constant
 size_t strlcat (char *, const char *, size_t);
                                             ^
In file included from recv.c:85:0:
direwolf.h:294:65: error: expected declaration specifiers or '...' before '__func__'
 #define strlcat(dst,src,siz) strlcat_debug(dst,src,siz,__FILE__,__func__,__LINE__)
                                                                 ^
/usr/include/string.h:83:45: error: expected declaration specifiers or '...' before numeric constant
 size_t strlcat (char *, const char *, size_t);
                                             ^
/usr/include/string.h:84:45: error: expected declaration specifiers or '...' before string constant
 size_t strlcpy (char *, const char *, size_t);
                                             ^
direwolf.h:293:65: error: expected declaration specifiers or '...' before '__func__'
 #define strlcpy(dst,src,siz) strlcpy_debug(dst,src,siz,__FILE__,__func__,__LINE__)
                                                                 ^
/usr/include/string.h:84:45: error: expected declaration specifiers or '...' before numeric constant
 size_t strlcpy (char *, const char *, size_t);
                                             ^
In file included from /usr/include/fortify/stdio.h:23:0,
                 from config.c:39:
/usr/include/fortify/string.h:156:1: error: 'strlcat' undeclared here (not in a function)
 _FORTIFY_FN(strlcat) size_t strlcat(char *__d, const char *__s, size_t __n)
 ^
In file included from config.c:43:0:
/usr/include/fortify/string.h: In function 'strlcat':
/usr/include/fortify/string.h:162:9: error: called object '__orig_strlcat' is not a function or function pointer
  return __orig_strlcat(__d, __s, __n);
         ^~~~~~~~~~~~~~
In file included from /usr/include/fortify/stdio.h:23:0,
                 from config.c:39:
/usr/include/fortify/string.h:156:1: note: declared here
 _FORTIFY_FN(strlcat) size_t strlcat(char *__d, const char *__s, size_t __n)
 ^
/usr/include/fortify/string.h: At top level:
/usr/include/fortify/string.h:165:1: error: 'strlcpy' undeclared here (not in a function)
[... 166 lines cut ...]
/usr/include/fortify/string.h:162:9: error: called object '__orig_strlcat' is not a function or function pointer
  return __orig_strlcat(__d, __s, __n);
         ^~~~~~~~~~~~~~
In file included from /usr/include/fortify/stdlib.h:28:0,
                 from dsp.c:31:
/usr/include/fortify/string.h:156:1: note: declared here
 _FORTIFY_FN(strlcat) size_t strlcat(char *__d, const char *__s, size_t __n)
 ^
/usr/include/fortify/string.h: At top level:
/usr/include/fortify/string.h:165:1: error: 'strlcpy' undeclared here (not in a function)
 _FORTIFY_FN(strlcpy) size_t strlcpy(char *__d, const char *__s, size_t __n)
 ^
In file included from dsp.c:35:0:
/usr/include/fortify/string.h: In function 'strlcpy':
/usr/include/fortify/string.h:171:9: error: called object '__orig_strlcpy' is not a function or function pointer
  return __orig_strlcpy(__d, __s, __n);
         ^~~~~~~~~~~~~~
In file included from /usr/include/fortify/stdlib.h:28:0,
                 from dsp.c:31:
/usr/include/fortify/string.h:165:1: note: declared here
 _FORTIFY_FN(strlcpy) size_t strlcpy(char *__d, const char *__s, size_t __n)
 ^
make: *** [<builtin>: config.o] Error 1
make: *** [Makefile.linux:365: gen_fff] Error 1
```

sgub, 2018-05-24:
Command to run alpine linux 3.7 in docker/raspbian:
# docker pull alpine && docker run -it alpine

My purpose to build direwolf in alpine linux is try to build a small (ideally < 50MB) image to run direwolf on any dockerized raspberry pi platform. The docker image size of rtl-sdr and direwolf based on resin/respbian  > 500MB, which is too big to run smoothly on various raspbeery platform.

wb2osz, 2018-05-24:
The traditional C library strcpy and strcat functions are very dangerous.  They have no checking for buffer overflow.  It is easy to write beyond the amount of allocated storage and stomp on other variables.  

strlcpy and strlcat have an extra argument for the destination length and perform checking to avoid buffer overflow.  BSD and Mac OS X have these functions.  For operating systems that don't have these functions, I provide my own.   Look in direwolf.h.

#if defined(__OpenBSD__) || defined(__FreeBSD__) || defined(__APPLE__)
// strlcpy and strlcat should be in string.h and the C library.
#else   // Use our own copy
....  (stuff for own version here)  ...

The solution is probably adding another symbol to the #if test condition.  The compiler probably has a predefined macro to identify the target platform.   Try this:    

 gcc -E -dM - < /dev/null

Do you see something resembling "alpine" or "musl" (for the C library version)?  Or maybe /usr/include/string.h has some definition to indicate whether strlcpy and strlcat are provided in this version of the library.

dranch, 2018-05-24:
John,
Are you saying that since this user is trying to build things in a container'ed version of Alpine Linux, your Makefile isn't able to identify the operating system and thus it's trying to use your internal copy of strcpy?

--David

wb2osz, 2018-05-24:
Yes, sort of.   Try this:

gcc -E -dM - < /dev/null

If this is done on Windows (with MinGW compiler), we find these among the macros predefined:

#define _WIN32 1
#define __WIN32 1
#define __WINNT 1
#define __WINNT__ 1
#define __WIN32__ 1
#define WIN32 1
#define WINNT 1

On Ubuntu,

#define __unix__ 1
#define __linux 1
#define __unix 1
#define __linux__ 1
#define __gnu_linux__ 1
#define unix 1
#define linux 1

Applications can use these to behave differently depending on the target platform.

If we were to do the same thing inside of a docker container, we would get different results based on the version of the compiler installed in the docker image, not on the host system.  It might be possible to use something predefined by the compiler.

Alternative 2:  The Makefile could be enhanced to find the OS flavor ( probably from /etc/os-release ) and add another definition to CFLAGS to indicate that the target environment already has strlcpy and strlcat.

I tried running Alpine:3.7 on Ubuntu because I already had docker there.

# grep Alpine /etc/*
/etc/issue:Welcome to Alpine Linux 3.7
/etc/motd:Welcome to Alpine!
/etc/motd:The Alpine Wiki contains a large amount of how-to guides and general
/etc/motd:information about administrating Alpine systems.
/etc/os-release:NAME="Alpine Linux"
/etc/os-release:PRETTY_NAME="Alpine Linux v3.7"

Alternative 3:  I noticed that Alpine Linux uses the "musl" C library.   Maybe it defines some symbol that we could use.

# apk update
# apk add build-base

Nothing immediately obvious.

Or maybe Makefile.linux could grep for strlcpy in /usr/include/string.h.

As a temporary work-around, I changed this part of direwolf.h

#if defined(OpenBSD) || defined(FreeBSD) || defined(APPLE)
// strlcpy and strlcat should be in string.h and the C library.

to be simply

#if 1

We now have another roadblock.   libasound2 doesn't seem to be available.

vielmetti, 2020-01-05:
I'm not 100% sure, but it looks like libasound2 on Alpine is called `alsa-lib`, based on https://pkgs.alpinelinux.org/package/edge/main/x86_64/alsa-lib

ericonr, 2020-12-28:
Now that the build system has switched to Cmake, it should be possible to add a simple check in it for the presence of `strlcat` and `strlcpy`, instead of relying on knowledge of particular properties of libcs. The current `#ifdef` doesn't cover NetBSD or DragonFlyBSD either, which both have `strl{cat,cpy}` available.

We've just hit the same error in Void Linux: https://github.com/void-linux/void-packages/pull/27510

wb2osz, 2020-12-31:
Would someone like to submit a proposed fix for this?

wb2osz, 2020-12-31:
I think I have a solution for the strlcpy/strlcat issue.
Please try the latest on the 'dev' branch.

Ma27, 2023-08-20:
FYI strlcat/strlcpy are available on glibc >=2.38. In case somebody has the same issue, here's a quick&dirty workaround: https://github.com/NixOS/nixpkgs/pull/247401/commits/a09911324cc8d59a9be91f62a4d857e1f52f5bc4

wb2osz, 2023-08-20:
Do you have a specific fix for the dev branch that won't break other platforms?
The dev branch will soon become release 1.7.

Ma27, 2023-09-03:
Not really, sorry. We're moving to glibc 2.38, so we don't have older glibc releases that don't support this. Also, I think there aren't many (widely-used) platforms left (BSDs Darwin included seem to support this for instance), right?

MarcinWieczorek, 2023-09-12:
`#define DEBUG_STRL 0` fixes build on ArchLinux (direwolf-git AUR package)

wb2osz, 2023-09-23:
There are multiple open cases for the same issue of building on Alpine.
These are combined into https://github.com/wb2osz/direwolf/issues/344 .
Closing this one.

closed 2023-09-23 by wb2osz