wb2osz/direwolf #152: PTT Releases too Early

MrSchumck, 2018-06-30:
So I've been experimenting with direwolf on my raspberry pi. I have PTT enabled on my GPIO pins however it seems like the bulk of the message is transmitted after the PTT disengages. this problem makes the push to talk function completely useless. Is there any way to increase the transmit time?

Thanks,

dranch, 2018-06-30:
Direwolf's Github "issues" area is for reporting bugs or requesting enhancements.  Please close this issue, join the Yahoo support group at direwolf_packet@yahoogroups.com, and repost your question.  When you port there, please include your Direwolf config, which GPIO pin you're using, and tell us if your using Direwolf for APRS, Serial KISS, AGW support, etc.

--David

MrSchumck, 2018-06-30:
The fact that the PTT doesn't stay down for the whole transmission is a bug as far as I can tell. Plus I am requesting an enhancement: I want the ability to edit the PTT time. I'll post the config file on here once I get home.

dranch, 2018-06-30:
Many people are having good luck with Direwolf so I imagine your issue is specific to your setup.  This issue system is NOT for troubleshooting.  Please close this Github issue and post your specific details on the Yahoo group

wb2osz, 2018-06-30:
What do you mean by "edit the PTT time?"    Have you read the **User Guide**, section 9.2.12, "Radio Channel – Transmit timing?"    

 https://github.com/wb2osz/direwolf/raw/dev/doc/User-Guide.pdf   

As David pointed out, additional details would be needed to try to reproduce the unexpected behavior.  

It might also be helpful to run direwolf with the "-d oo" command line option to print out information about PTT being turned on and off.

closed 2018-07-01 by MrSchumck

na7q, 2018-07-01:
What you want is a longer TXTAIL. Not sure on the length you need or the setup you have. But the TXTAIL setting would keep the transmit up a bit longer at the tail of the packets.
Only issue I've experienced with the GPIO pins is sometimes they like to get stuck in the "HIGH" position. Leaving the radio transmitting until it hits the time out timer limit.