wb2osz/direwolf #156: v1.3 OSX - Floating point exception: 8

sm3ulc, 2018-07-09:
Testing the 1.3 binary version for OSX. Runs for a some minutes as igate before dying with last logging:

rtl_fm -f 144.8M - | ./direwolf -c sdr.conf -t 0 -n 1 -r 24000 -D 1 -l . -
....
Floating point exception: 8

wb2osz, 2018-07-09:
That's not much to go on.  Please provide a copy of everything on the screen leading up to the error.  Perhaps there was some specific data format or value which caused the problem.

sm3ulc, 2018-07-10:
Found bug again. Running aprsis32 against the agw-port. Easily repeated. Though possible better to focus on getting the latest beta-dev running.

time rtl_fm -f 144.8M - | ./direwolf -c sdr.conf -t 0 -d i -n 1 -r 24000 -D 1 -l . -
Dire Wolf version 1.3 (Mar 19 2016) Beta Test
Warning: Could not open 'symbols-new.txt'.
The "new" overlayed character information will not be available.

Reading config file sdr.conf
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
2018-07-10 17:41:23.808 direwolf[31762:4658908] Error loading /Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter:  dlopen(/Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter, 262): Library not loaded: /System/Library/Frameworks/Jackmp.framework/Versions/A/Jackmp
  Referenced from: /Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter
  Reason: image not found
2018-07-10 17:41:23.808 direwolf[31762:4658908] Cannot find function pointer New_JackRouterPlugIn for factory 7CB18864-927D-48B5-904C-CCFBCFBC7ADD in CFBundle/CFPlugIn 0x7941a250 </Library/Audio/Plug-Ins/HAL/JackRouter.plugin> (bundle, not loaded)
Audio input device for receive: stdin  (channel 0)
Audio out device for transmit:   (channel 0)
Audio input buffer has unexpected extreme size of 0 bytes.
Detected at audio_portaudio.c, line 723.
This might be caused by unusual audio device configuration values.
Using 2048 to attempt recovery.
Audio output buffer has unexpected extreme size of 0 bytes.
Detected at audio_portaudio.c, line 732.
This might be caused by unusual audio device configuration values.
Using 2048 to attempt recovery.
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 24000 sample rate.
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Use -p command line option to enable KISS pseudo terminal.
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS client application on port 8001 ...
Found Rafael Micro R820T tuner

Connected to AGW client application 0...

Ready to accept AGW client application 1 on port 8000 ...
Tuner gain set to automatic.
Tuned to 145052000 Hz.
Oversampling input by: 42x.
Oversampling output by: 1x.
Buffer size: 8.13ms
Exact sample rate is: 1008000.009613 Hz
Sampling at 1008000 S/s.
Output at 24000 Hz.

SM0ULC-9 audio level = 42(10/11)   [NONE]   _|||||||_
[0.4] SM0ULC-9>VQ2U01,WIDE1-1,WIDE2-1:`,6*l"9>\"5/}bat=0.51V bus=3.27V T=38.4C GPS=12
MIC-E, OVERLAYED CAR, Unknown manufacturer, En Route
N 61 25.0100, E 016 26.1400, 0 MPH, course 229, alt 377 ft
bat=0.51V bus=3.27V T=38.4C GPS=12
Opening log file "2018-07-10.log".

Now connected to IGate server euro.aprs2.net (80.89.46.248)
Check server status here http://80.89.46.248:14501

[rx>ig] user SM3ULC pass 24505 vers Dire-Wolf 1.3
[ig] # aprsc 2.1.4-g408ed49
[ig] # logresp SM3ULC verified, server T2NORWAY

Error getting message header from AGW client application 0.
Tried to read 36 bytes but got only 0.
Closing connection.

[ig] SM3ULC-1>APDW13:!6125.01NR01626.18E&PHG0200Sjorgra
[rx>ig] SM3ULC-1>APDW13,qAR,SM3ULC-1:!6125.01NR01626.18E&PHG0200Sjorgra

SM0ULC-9 audio level = 41(11/9)   [NONE]   _|||||||_
[0.4] SM0ULC-9>VQ2U01,WIDE1-1,WIDE2-1:`,6*l!]>\"5/}bat=0.51V bus=3.27V T=38.2C GPS=12
MIC-E, OVERLAYED CAR, Unknown manufacturer, En Route
N 61 25.0100, E 016 26.1400, 0 MPH, course 165, alt 377 ft
bat=0.51V bus=3.27V T=38.2C GPS=12
[rx>ig] SM0ULC-9>VQ2U01,WIDE1-1,WIDE2-1,qAR,SM3ULC-1:`,6*l!]>\"5/}bat=0.51V bus=3.27V T=38.2C GPS=12

Connected to AGW client application 1...

Ready to accept AGW client application 0 on port 8000 ...
[0L] SM3ULC>APWW10,WIDE1-1,WIDE2-1:@154202h6124.99N/01626.17ElAPRS-IS for Win32
Signal caught, exiting!

User cancel, exiting...
Signal caught, exiting!
Floating point exception: 8

wb2osz, 2018-07-11:
There are other troubling messages:

Warning: Could not open 'symbols-new.txt'.
The "new" overlayed character information will not be available.

Error loading /Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter: dlopen(/Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter, 262): Library not loaded: /System/Library/Frameworks/Jackmp.framework/Versions/A/Jackmp
Referenced from: /Library/Audio/Plug-Ins/HAL/JackRouter.plugin/Contents/MacOS/JackRouter
Reason: image not found

Audio input buffer has unexpected extreme size of 0 bytes.
Detected at audio_portaudio.c, line 723.
This might be caused by unusual audio device configuration values.
Using 2048 to attempt recovery.
Audio output buffer has unexpected extreme size of 0 bytes.
Detected at audio_portaudio.c, line 732.
This might be caused by unusual audio device configuration values.
Using 2048 to attempt recovery.

That software version is more than 2 years old.  There have been many improvements and bug fixes since then.

Trying the most recent version would be a better use of time.