wb2osz/direwolf #157: Position Ambiguity using GPS

LasersAreGreat, 2018-07-10:
Hello.  Firstly, let me say that direwolf is awesome!  I've setup a couple of igates in the last week with it and couldn't be happier.  Thank you for all the work you've put into it!

One of the igates I setup is a mobile igate in my car, running on a raspberry pi and using the onboard wifi to igate packets that are received. I want the igate's rough position to be shared, but I would rather not share my exact position.  I know that position ambituity is the solution here, but can't seem to find that option for tbeacon.  Is there any way to add position ambiguity?  If not, can that please be considered as a feature request?

Thank you.

LasersAreGreat, 2018-07-21:
I should clarify, I tried the ambiguity option (and the short "ambig") with tbeacon, but it throws an error. It seems that for TBEACON, ambiguity is not an option

wb2osz, 2018-07-22:
What do you mean by "throws an error?"   Please provide more details.  What is in the configuration file?  What do you see on the screen?

 This seems to work OK for me.

Configuration file:

mycall abcdef
gpsnmea com4
tbeacon delay=0:35 every=1 
tbeacon delay=0:40 every=1 ambiguity=1
tbeacon delay=0:45 every=1 ambig=2
tbeacon delay=0:50 every=1 ambig=3
tbeacon delay=0:55 every=1 ambig=4

Result:

[0L] ABCDEF>APDW15:!4237.13N/07120.84W-
[0L] ABCDEF>APDW15:!4237.1 N/07120.8 W-
[0L] ABCDEF>APDW15:!4237.  N/07120.  W-
[0L] ABCDEF>APDW15:!423 .  N/0712 .  W-
[0L] ABCDEF>APDW15:!42  .  N/071  .  W-

LasersAreGreat, 2018-07-22:
Here is the output I get when running direwol (I made the error line bold)f:

Reading config file /home/pi/direwolf/direwolf.conf
**Config file, line 488: Invalid option keyword, ambiguity.**
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS client application on port 8001 ...
Use -p command line option to enable KISS pseudo terminal.

The relevant line (488) of the config file is this:

TBEACON sendto=IG delay=1 every=5 ambiguity=1 overlay=S symbol="igate" COMMENT="RX-only mobile igate"

I get the same error whether I use TBEACON or IBEACON

wb2osz, 2018-07-22:
What version are you using?   From the revision history, it looks like this new option was added June 2017, after version 1.4.   In other words, you need version 1.5.

closed 2018-07-22 by wb2osz