wb2osz/direwolf #159: Feature Request: Quiet IG->TX Output

nayrnet, 2018-07-13:
I'm piping STDOUT into [frontail](https://github.com/mthenw/frontail) to create a simple WebUI to display log files.. This is mostly for other RF stations nearby to use to help them tune outputs and paths.. basically letting em see what my iGate can see.

However all the IG->TX Lines are undesired, I commented out the code and rebuilt it.. but I think it'd be better to have a CLI option to silence these lines of the logs.. I tried filtering out what I didnt want from the upstream server but got very inconsistent results in what IS servers support what.. sometimes I'd get filtered, sometimes all filters would be ignored and it'd be even worse.

wb2osz, 2018-07-14:
I like your IGate status page at  http://igate.nayr.net/   You should mention this in the Dire Wolf and other APRS discussion groups.  I'm sure a lot of people would be interested.

nayrnet, 2018-07-14:
done and done, eventually I'll create a Github repo w/all my configs and a nice readme on how to use it.. perhaps a script to set it all up.

I went ahead and added the feature mimicking your style and submitted a merge request, but I'm sure you've already seen that.

closed 2018-07-14 by nayrnet

TheButterZone, 2019-03-14:
Any way to quiet ig>tx from the SSH...
$ sudo journalctl -o short -af -u direwolf