wb2osz/direwolf #164: Feature Request: Stats File

nayrnet, 2018-07-21:
Story: As an operator I'd like an easy way to scrape internal stats for use with external graphing (MRTG)

Proposal: I've implemented a new beacon type called SBEACON that replicates the functionality of IBEACON but only writes to /tmp/dw-stats.txt

```
                case BEACON_STATS:

                  {
                    int last_minutes = 60;

                    FILE *statFile = fopen("/tmp/dw-stats.txt", "w");
                    if (statFile == NULL)
                        {
                            dw_printf("Error opening stats file!\n");
                        }

                    fprintf (statFile, "MSG_CNT=%d\r\nPKT_CNT=%d\r\nDIR_CNT=%d\r\nLOC_CNT=%d\r\nRF_CNT=%d\r\nUPL_CNT=%d\r\nDNL_CNT=%d\r\n",
                                                igate_get_msg_cnt(),
                                                igate_get_pkt_cnt(),
                                                mheard_count(0,last_minutes),
                                                mheard_count(g_igate_config_p->max_digi_hops,last_minutes),
                                                mheard_count(8,last_minutes),
                                                igate_get_upl_cnt(),
                                                igate_get_dnl_cnt());
                    fclose(statFile);
                    strlcpy (beacon_text, "", sizeof(beacon_text));  // abort!
                  }
                  break;
```

I set this to beacon every 5mins and then I whipped up a few simple bash scripts to feed MRTG

```
#!/bin/bash
# MRTG Script for DireWolf

source /tmp/dw-stats.txt
echo $DNL_CNT
echo $UPL_CNT
uptime -p
hostname -a
```
and
```
#!/bin/bash
# MRTG Script for DireWolf

source /tmp/dw-stats.txt
echo $DIR_CNT
echo $LOC_CNT
uptime -p
hostname -a
```

You can see them in action at http://igate.nayr.net

I'd offer a merge request, but my solution is not windows compatible (hardcoded filename) and kinda stomps on the toes of beacons with this pseduo fake beacon.. so I dont think you'll take it.. But I would like to discuss such functionality.

dm-o, 2018-09-28:
Sound very interesting. I love to give out stats on my own website too.

bhcleek, 2021-04-14:
I wonder if this feature might be accepted by leveraging the IBEACON, but using a file path in the VIA field to cause it to write to a file similarly to how NOGATE will cause it to only write to the usual log file. Using VIA with a file path value could allow the beacon stats to be written to a specific file.

@wb2osz would you take such a patch?

wb2osz, 2021-04-14:
There are a few outstanding requests for something external to change direwolf parameters on the fly or to query the internal state, e,g,  statistics, recently heard stations, and so on.
Rather polling for lots of assorted files, I was thinking of a more unified approach such as a network interface.  Not necessarily REST, but that concept.

bhcleek, 2021-04-14:
That's sounds great. I was just thinking last night that I wish I could reload the config with restarting. SIGHUP support would be a welcome change, too