wb2osz/direwolf #165: Direwolf 1.4: "Audio output failure waiting for buffer."

kd8drx, 2018-08-05:
**Direwolf Version:** 1.4
**OS:** Windows 7 x64 SP1
**Hardware:** VMWare Fusion 10 on a Macbook Pro
**Audio:** Signalink USB connected directly into VMWare
**Modem Speed:** 1200 Baud

**Issue:**
When running Direwolf, the ambiguous error "Audio output failure waiting for buffer." will sometimes appear. This generally happens shortly after connection to a station, usually less than a minute after starting Direwolf itself. Once the error occurs, audio produced by Direwolf is heavily distorted and is indecipherable by the remote station. This persists until Direwolf is killed and restarted.

The audio itself sounds like a 1200 Baud AFSK signal, but speed up roughly 2-4x what it should be, at a slightly higher pitch. I'm happy to provide a recording if that'd be helpful.

The code itself is at https://github.com/wb2osz/direwolf/blob/master/audio_win.c#L936, though not being a C Programmer I have _zero_ idea what it's doing. This appears related to (or the same as) issue #78.

Direwolf Output at error time:
```
KC6SSM-10 audio level = 51(19/22)   [NONE]   ||||||___
[0.2] KC6SSM-10>KD8DRX:(I cmd, n(s)=3, n(r)=0, p=1, pid=0xf0)CMS via KC6SSM ><0x0d>
[0L] KD8DRX>KC6SSM-10:(RR res, n(r)=4, f=1)
[0L] KD8DRX>KC6SSM-10:(I cmd, n(s)=0, n(r)=4, p=0, pid=0xf0);FW: KD8DRX<0x0d>
[0L] KD8DRX>KC6SSM-10:(I cmd, n(s)=1, n(r)=4, p=0, pid=0xf0)[RMS Express-1.5.13.0-B2FHM$<0x0d>
[0L] KD8DRX>KC6SSM-10:(I cmd, n(s)=2, n(r)=4, p=0, pid=0xf0);PR: 10490210<0x0d>
Audio output failure waiting for buffer.
[0L] KD8DRX>KC6SSM-10:(I cmd, n(s)=3, n(r)=4, p=0, pid=0xf0); KC6SSM-10 DE KD8DR
X (CM87TS)<0x0d>
[0L] KD8DRX>KC6SSM-10:(DISC cmd, p=1)
```

dranch, 2018-08-06:
I would generally say that running anything under emulation (Vmware Fusion) might cause performance issues as there are so many more possible variables for failure.

Are your versions of OSX, VM Fusion, and Windows 7 the *current* versions and also fully patched?  Please also install Direwolf v1.5Beta2 and see if you can still reproduce your issue.  Can you connect and route a dedicated USB soundcard ($8 Syba units are well supported) into the Windows domain (not seen in the OSX domain) and see if you can reproduce the issue?

--David

wb2osz, 2019-11-24:
This was two versions back.
I'm also skeptical about using audio in an application running under VMware.
Reopen if still a problem.

closed 2019-11-24 by wb2osz