wb2osz/direwolf #166: Direwolf running in background ontop of Pi-star

on2aav, 2018-08-10:
I would love to use 'any' kind of beacon within Direwolf, using GPS live position data, but only send it over IGATE.
All is working fine, ontop of the Pi-star hotspot software, the only thing i need to do is find a way to have the Normal beacon (car tracker) to be able to send it's location update over Igate and not to the normal audio output.
 Any beacon i tested all want a fixed GPS location as extra info.

All help is welcome of course ;)
73 ON2ON

dranch, 2018-08-10:
Github issues are for reporting bugs or enhancement requests.  Please close this issue and post your support question on the Yahoo Groups list at direwolf_packet@yahoogroups.com .  With that said, your request is possible though quirky as Direwolf really requires a soundcard to be present.

on2aav, 2018-08-10:
I will close the issue here, and to comment, the soundcard is not the issue.
Direwolf is working fine.

Best rgds
Luc

closed 2018-08-10 by on2aav