wb2osz/direwolf #167: not an issue, rather a question about going crazy cheap...

pleasantone, 2018-08-19:
John,

Sorry to use the issues interface, I did not find a good e-mail for you. Mine is my callsign at shockwave.org.

Thank you for all your work. I used to do a lot of packet work back in the 80s and just for giggles got interested in your take on using sound cards. I’m really impressed with where you have taken the software.

I noticed in the livestream of the MicroHAMS talk you gave, that you showed a USB sound dongle that had been hand-modified to create a PTT circuit by poking into a GPIO pin that was inside the dongle.

I was curious how well that worked for you?  If it’s suitable for 2m packet without causing a lot of ground noise, I was thinking of making up a batch of them for our local ARES along with some VirtualBox disk images running Linux w/direwolf pre-configured.  I wanted to try to make a “almost completely virtual” packet setup as an experiment.

In San Francisco, we have plenty of people with powerful laptops, but few of them are messing around with RPis et al.

Do you have any thoughts?

Best,

Paul
k6pst

dranch, 2018-08-19:
Hello Paul,

As you mentioned, the Github issues are is for real bugs or enhancement requests.  Please use direwolf_packet@yahoogroups.com for support issues.  To your question, the function of a modified  CM108 sound card works fine.  I've personally observed some background noise (bus noise) coming from the USB ground but that was an issue of the upstream computer and not the soundcard's fault.  It didn't impact the performance of the packet signal though.  Please close this issue and re-post your message on the YahooGroup if you have other questions.

--David
KI6ZHD

pleasantone, 2018-08-19:
Thank you both very much.

closed 2018-08-19 by pleasantone