wb2osz/direwolf #169: agw Y-frames

ve1jot, 2018-09-17:
I have been told that DW appears to be missing Y-frame flow control....

wb2osz, 2018-09-17:
According to the source code it is in version 1.5.   Please let me know if it is not functioning as expected.

ve1jot, 2018-09-18:
"it should return the number of frames waiting to be transmitted on the given AX.25 connection, however it always returns zero"

ve1jot, 2018-09-18:
"EasyTerm keeps the frame queue of 20 frames.

So if we have a large amount of data, EasyTerm transmits first 20 frames and waiting for Y-frame result.

In decreasing order of the queue, it will add the new frames and keep the number of 20."

cgarrett118, 2018-10-07:
ve1jot Try direwolf 1.5 beta4. I do not get any errors for the agw port.
Chester KD9JLZ

ve1jot, 2019-01-04:
thanks Chester, but I wasn't getting actual errors, just odd behaviour with yapp file transfers...I will close this issue for now until I have more information and somehow narrow it down..

wb2osz, 2019-01-04:
I have a fix for this but am not quite ready to release it.