wb2osz/direwolf #170: random TXDelay timing, Direwolf 1.5-beta4

meinja, 2018-09-24:
I have Direwolf 1.5-beta4 installed on a Raspberry Pi3 ( with freshly installed Raspian Stretch).  The setup uses linBPQ and Direwolf KISS mode 127.0.0.1:8001.  I am able to make connections with no problem on 1200 and 300 baud, but I notice random delays between the PTT activation and actual transmitted audio.  I have TXDelay set at 30ms, but I'm seeing delays up to 2 seconds (maybe even more at times) between PTT keydown and transmitted audio.  Seems purely random; I would say 50% of the transmissions have this large delay, and the rest are normal.  The long delays themselves are random in length also, and occur on any transmit, whether an ack or a command sent to the other station.  The Pi3's CPU is generally around only 3% use when I'm using packet; no spikes during the delays or anything.

PTT is activated by the Pi3's GPIO pin 26.  

USB audio is the Syba SD-CM-UAUD.

Timings in bpq32.cfg are at the suggested settings in the Direwolf user guide.

The BPQ config file is very basic; no other timing settings other than the following:

PORT
   PORTNUM=1                    ; Port number
   ID=VHF Packet 1200-baud      ; PORTS command text
   TYPE=ASYNC                  :RS232 connection
   IPADDR=127.0.0.1         ; DIREWOLF
   TCPPORT=8001             ; DIREWOLF
   SPEED=19200
   CHANNEL=A                   ; TNC channel
   MAXFRAME=1               ;Max outstanding frames
   FRACK=4000                 ; Level 2 timeout (ms)
   RESPTIME=40                    ; Level 2 delayed ACK (ms)
   RETRIES=10                         ; Level 2 max retries
   PACLEN=128                     ; Max packet length (bytes)                                     
   TXDELAY=100                    ; Transmit keyup delay (ms)
   SLOTTIME=100                  ; CMSA interval timer (ms)
   TXTAIL=100
   PERSIST=63                     ; Persistence (256/(# transmissions-1)
   DIGIFLAG=1                   ; Allow Digipeat on this port
ENDPORT

Direwolf config settings that are enabled are:

ADEVICE  plughw:1,0
ACHANNELS 1
MODEM 1200 1200:2200
PTT GPIO 26
KISSPORT 8001

I also tried different ARATE settings at 11025 and 48000, but no change was seen.

Output when I run it is:

Dire Wolf version 1.5 (Sep 22 2018) Beta Test 4
Includes optional support for:  cm108-ptt

Reading config file direwolf.conf
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS TCP client application 0 on port 8001 ...

Attached to KISS TCP client application 0...

Ready to accept KISS TCP client application 1 on port 8001 ...
KISS protocol set TXDELAY = 10 (*10mS units = 100 mS), port 0
KISS protocol set Persistence = 63, port 0
KISS protocol set SlotTime = 10 (*10mS units = 100 mS), port 0
KISS protocol set TXtail = 10 (*10mS units = 100 mS), port 0
KISS protocol set FullDuplex = 0, port 0

SmithChart, 2018-10-31:
I can reproduce this issue with the current master c0abb4b2164a288bc5292114bbfe196629b069ea on my raspberryPi zero W. I am using direwolf in beacon-mode and let it transmit my current system uptime as a dummy value.

I am using a GPIO to toggle the PPT.
My CPU-load is a bit higher (around 20%). But that does not justify a few seconds between assertion of the PPT-GPIO and the actual audio playing.

I have tried to narrow the delay down by looking at the log output:
```
pi@raspberrypi:~ $ direwolf -c direwolf.conf -d o | ts '[%Y-%m-%d %H:%M:%S]'
[2018-10-31 21:29:55] Dire Wolf version 1.5
[2018-10-31 21:29:55] 
[2018-10-31 21:29:55] Reading config file direwolf.conf
[2018-10-31 21:29:55] Audio device for both receive and transmit: plughw:1,0  (channel 0)
[2018-10-31 21:29:55] Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
[2018-10-31 21:29:55] Ready to accept KISS TCP client application 0 on port 8001 ...
[2018-10-31 21:29:55] Ready to accept AGW client application 0 on port 8000 ...
(...)
```

But observing roughly 20 TX-Packets they had no delay. The following log shows such a sample:
```
[2018-10-31 21:33:10] PTT 0 = 1
[2018-10-31 21:33:10] [0L] DB1FI>APDW15,WIDE2-1:T#59,4412.56
[2018-10-31 21:33:11] PTT 0 = 0
```

I have used an USB-Serial-Adapter to cross-check the problem: With a USB-Serial-Adapter (and a new config-file reflecting the change) the delay seems to be gone, too.

Once I've got the time I'll try to reproduce the issue again.

meinja, 2018-11-02:
Thank you very much for that information.  I will try that here tonight and see if it clears up the delay.

meinja, 2018-11-03:
Unfortunately, I still have the delay, after running Direwolf with "direwolf -c direwolf.conf -d o".

meinja, 2018-11-04:
> 
> 
> I can reproduce this issue with the current master [c0abb4b](https://github.com/wb2osz/direwolf/commit/c0abb4b2164a288bc5292114bbfe196629b069ea) on my raspberryPi zero W. I am using direwolf in beacon-mode and let it transmit my current system uptime as a dummy value.
> 
> I am using a GPIO to toggle the PPT.
> My CPU-load is a bit higher (around 20%). But that does not justify a few seconds between assertion of the PPT-GPIO and the actual audio playing.
> 
> I have tried to narrow the delay down by looking at the log output:
> 
> ```
> pi@raspberrypi:~ $ direwolf -c direwolf.conf -d o | ts '[%Y-%m-%d %H:%M:%S]'
> [2018-10-31 21:29:55] Dire Wolf version 1.5
> [2018-10-31 21:29:55] 
> [2018-10-31 21:29:55] Reading config file direwolf.conf
> [2018-10-31 21:29:55] Audio device for both receive and transmit: plughw:1,0  (channel 0)
> [2018-10-31 21:29:55] Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
> [2018-10-31 21:29:55] Ready to accept KISS TCP client application 0 on port 8001 ...
> [2018-10-31 21:29:55] Ready to accept AGW client application 0 on port 8000 ...
> (...)
> ```
> 
> But observing roughly 20 TX-Packets they had no delay. The following log shows such a sample:
> 
> ```
> [2018-10-31 21:33:10] PTT 0 = 1
> [2018-10-31 21:33:10] [0L] DB1FI>APDW15,WIDE2-1:T#59,4412.56
> [2018-10-31 21:33:11] PTT 0 = 0
> ```
> 
> I have used an USB-Serial-Adapter to cross-check the problem: With a USB-Serial-Adapter (and a new config-file reflecting the change) the delay seems to be gone, too.
> 
> Once I've got the time I'll try to reproduce the issue again.

If I'm understanding this right, when you run Direwolf without the "-d o" output switch, the TX delay issue is reproduced?  And if you run it with the "-d o" output switch, the TX delay issue is gone?  I tried it both ways here, but still have the TX delay problem.

SmithChart, 2018-11-05:
Sorry, that is a misunderstanding.
 `direwolf -d o ` just activates debug output for the PPT and carrier-detect signals. Thus I just get more output from direwolf and get some information about internal states. 

I am not really sure what caused the delay or what made it disappear when I started with debug output. @meinja  could you run direwolf with the command I used ( `direwolf -c direwolf.conf -d o | ts '[%Y-%m-%d %H:%M:%S]` and paste the output of a TX with a delay here? I am curious if we can reassemble the delay in the timestamps.

meinja, 2018-11-06:
OK,  I'll post the output later today.  What Linux distro are you using on the Pi Zero?  The "direwolf -c direwolf.conf -d o | ts '[%Y-%m-%d %H:%M:%S]'" command doesn't work on my Raspian Stretch, I get an error saying "no command found" for "ts".   I ran "direwolf -c direwolf.conf -d o -T %H:%M:%S" which did work on mine.  Very curious what you did to eliminate the delay.  I did notice there is no apparent delay when using 300-baud, which I thought was odd.  I'll have to test that some more to verify though.

meinja, 2018-11-06:
Nice!  Thanks.  Here's the output.  I made three transmissions to my other setup across the house by just sending the letter "a".   The lines in bold are the transmits.  First transmit was noticeably much longer with the delay.  Second and third were normal:

[2018-11-06 17:27:23] PTT 0 = 1
**[2018-11-06 17:27:23] [0L] KP3FT-2>KB2QXA:(I cmd, n(s)=3, n(r)=6, p=1, pid=0xf0)a<0x0d>**
[2018-11-06 17:27:24] PTT 0 = 0
[2018-11-06 17:27:24] DCD 0 = 1
[2018-11-06 17:27:24] DCD 0 = 0
[2018-11-06 17:27:24] DCD 0 = 1
[2018-11-06 17:27:24] 
[2018-11-06 17:27:24] KB2QXA audio level = 70(31/16)   [NONE]   _|||||||_
[2018-11-06 17:27:24] [0.4] KB2QXA>KP3FT-2:(RR res, n(r)=4, f=1)
[2018-11-06 17:27:25] DCD 0 = 0
[2018-11-06 17:27:27] PTT 0 = 1
**[2018-11-06 17:27:27] [0L] KP3FT-2>KB2QXA:(I cmd, n(s)=4, n(r)=6, p=1, pid=0xf0)a<0x0d>**
[2018-11-06 17:27:27] PTT 0 = 0
[2018-11-06 17:27:28] DCD 0 = 1
[2018-11-06 17:27:28] 
[2018-11-06 17:27:28] KB2QXA audio level = 68(31/16)   [NONE]   _||||||||
[2018-11-06 17:27:28] [0.4] KB2QXA>KP3FT-2:(RR res, n(r)=5, f=1)
[2018-11-06 17:27:28] DCD 0 = 0
[2018-11-06 17:27:29] PTT 0 = 1
**[2018-11-06 17:27:29] [0L] KP3FT-2>KB2QXA:(I cmd, n(s)=5, n(r)=6, p=1, pid=0xf0)a<0x0d>**
[2018-11-06 17:27:30] PTT 0 = 0
[2018-11-06 17:27:30] DCD 0 = 1

SmithChart, 2018-11-08:
Can you estimate how long _noticeably much longer_ is? In the log the TX have the following duration:
1) <2s
2) <1s
3) <2s

I will try to reproduce this with my setup today.

meinja, 2018-11-08:
That's very interesting.  I wonder if it's related to amperage somehow.  Is your USB hub an externally-powered hub, or "passive" using the Pi Zero's power?  

The noticeable delays were around 2 seconds, and seems like 3 seconds at times.   Some are around 1 second.  On mine transmissions with normal delay sounds like just a quick chirp on the back-and-forth acks and short one-letter commands.

meinja, 2018-11-12:
I found this in the "xmit.c" file in the Direwolf directory around line# 882:

* See how long it has been since PTT was turned on.
 * Wait additional time if necessary.
 */

	time_now = dtime_now();
	already = (int) ((time_now - time_ptt) * 100.);
	wait_more = duration - already;

wb2osz, 2019-02-16:
This has been isolated to a small test case.
https://github.com/wb2osz/rpi-usb-audio-bug

meinja, 2019-02-26:
Has anyone tried an older version Raspian distro to see if that makes a difference?  I'll try it here on my RPi3b, but it won't be for some time yet.

wb2osz, 2019-11-24:
The problem was reported here but no one seems interested.
https://bugs.launchpad.net/bugs/1819560
It was confirmed as being a problem but it hasn't been assigned to anyone.
Maybe if more people piled on to the comment section it would get some attention.

wb2osz, 2019-11-24:
Same as https://github.com/wb2osz/direwolf/issues/194

wb2osz, 2023-10-20:
Duplicate.  See   https://github.com/wb2osz/direwolf/issues/194

closed 2023-10-20 by wb2osz