wb2osz/direwolf #176: gpio permissions on raspberry

IW3IJQ, 2018-10-24:
I have direwolf running on raspberry pi with user direwolf. User is part of gpio group.
When I run direwolf this happens:

 

```
direwolf@raspberrypi:~ $ direwolf -a 100 -doo
Dire Wolf version 1.5
Includes optional support for:  gpsd cm108-ptt

Reading config file direwolf.conf
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
ch=0, PTT method=2, device=, line=0, gpio=21, lpt_bit=0, invert=0
ch=0, DCD method=2, device=, line=0, gpio=20, lpt_bit=0, invert=0
ch=0, CON method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=1, PTT method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=1, DCD method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=1, CON method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=2, PTT method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=2, DCD method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=2, CON method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=3, PTT method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=3, DCD method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=3, CON method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=4, PTT method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=4, DCD method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=4, CON method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=5, PTT method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=5, DCD method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
ch=5, CON method=0, device=, line=0, gpio=0, lpt_bit=0, invert=0
/sys/class/gpio/export: uid=0, gid=997, mode=o100770
my uid=1001, gid=1001, supplementary groups= 20 29 997 1001
My supplemental group 997 matches and we have group write permission.
/sys/class/gpio/export: uid=0, gid=997, mode=o100770
my uid=1001, gid=1001, supplementary groups= 20 29 997 1001
My supplemental group 997 matches and we have group write permission.
Contents of /sys/class/gpio:
        .
        ..
        export
        gpio21
        gpiochip0
        gpiochip128
        unexport
Path for gpio number 21 is /sys/class/gpio/gpio21
/sys/class/gpio/gpio21/direction: uid=0, gid=0, mode=o100644
my uid=1001, gid=1001, supplementary groups= 20 29 997 1001
Trying 'sudo chmod go+rw /sys/class/gpio/gpio21/direction' hack.

Questa lezione dovrebbe essere stata impartita dall'amministratore
di sistema locale. Solitamente equivale a:

    #1) Rispettare la privacy degli altri
    #2) Pensare prima di digitare
    #3) Da grandi poteri derivano grandi responsabilità

[sudo] password di direwolf:
You don't have the necessary permission to access GPIO.
There are three different solutions:
 1. Run as root. (not recommended)
 2. If operating system has 'gpio' group, add your user id to it.
 3. Configure your user id for sudo without a password.

Read the documentation and try -doo command line option for debugging details.

direwolf@raspberrypi:~ $ ls -la /sys/class/gpio/gpio21/direction
-rwxrwx--- 1 root gpio 4096 ott 24 12:55 /sys/class/gpio/gpio21/direction
```
direwolf write 21 in /sys/class/gpio/export and check permission on /sys/class/gpio/gpio21/direction.
It hasn't permission to write and try to use sudo.
Terminating direwolf and checking  /sys/class/gpio/gpio21/direction the permissions are ok.

This because udev take a little time to adjust permissions after enabling gpio.
This can be solved with a little delay (one second) after write to  /sys/class/gpio/export.

wb2osz, 2019-07-07:
This is now in the "dev" branch and will be in release 1,6.
Thanks for submitting this improvement.

closed 2019-07-07 by wb2osz

craigerl, 2021-07-05:
Problem observed with direwolf 1.6

```
pi@gtown:~ $ tail -f /run/direwolf.log 
Line 8: Invalid port number for IGate server. Using default 14580.
Audio device for both receive and transmit: plughw:0,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
You don't have the necessary permission to access GPIO.
There are three different solutions: 
 1. Run as root. (not recommended)
 2. If operating system has 'gpio' group, add your user id to it.
 3. Configure your user id for sudo without a password.

Read the documentation and try -doo command line option for debugging details.
```

```
Dire Wolf version 1.6                                                                                                           
```