wb2osz/direwolf #183: dw stops collecting packets after 2-6 hrs, needs restart

vk2auj, 2018-12-08:
All working fine in the first 2 to 6 hrs after startup, then eventually, dw still beaconing but packets are not longer collected and/or sent to the internet.
A workaround for now is a timer that shuts down the device for 1 min every hour.
Raspberry PI 3, wifi, RTL dongle.
See attached a ps- al done:
- whilst is no longer collects packet;
- then after restart.
[image]
[image]

dranch, 2018-12-08:
We need more details about your setup. What version of Raspbian?  Is it fully patched?  Which version of Direwolf?  What version of rtl_fm?  You should be running the newest of all of these to really be able to help you as no one else has reported an issue like this.  When you start Direwolf, where are you sending the logs?  Can you send a tgz of the logs lines from the transition when it's working to when it starts to fail?  Can you also start Direwolf with the strace command and log to a USB pen drive or something large and send us just some examples of the longs when it's failed?  You'll have to prune that file WAY back as it will be extremely huge.

--David
KI6ZHD

vk2auj, 2018-12-08:
Thank you David for the reply.  The only thing I could find out myself was the version of Raspbian?  version 9 (Stretch).

As for patch level, Direwolf and rtf_fm version, I couldn't find a way on the net on how to extract the information.

Yves vk2auj

dranch, 2018-12-09:
When you start Direwolf, it will print out it's version to STDOUT.  If you're redirecting that output to a log file, look in there.  As for the rtl_fm program, that's not clear to me.  I have an old version here.. 0.5.2 which doesn't have a command line option to give the version nor does it seem to be found using the "strings" program.  How did you install it?  If you installed it via a package, you can run "dpkg -l | grep -i rtl" and see if it's listed there.

vk2auj, 2018-12-09:
David.  Hold on ... it was suggested it could be a PSU issue, so I've disconnected the TV and USB dongle for wireless keyboard and will let it running to test.  I'll also try to find a 3amp PSU.

vk2auj, 2018-12-14:
After investigations, it appeared that the RTL dongle was getting too hot and one of the processor was probably shutting itself off.  By simply removing the dongle from the metal case, I was a able to do an uninterrupted run of 28hrs, and then 40hrs and still running  after relocating to its permanent location.  I'll try to design a heatsink for the regulator and the 2 main chips of the RTL dongle.

closed 2018-12-14 by vk2auj

dranch, 2018-12-15:
Which RTL dongle do you have?  Instead of trying to retrofit something that probably won't go well, buy something better.  On the lower end, consider say something like:

   https://www.nooelec.com/store/sdr/sdr-receivers/nesdr/nesdr-smart-sdr.html

Or if you want something with a LOT better receiver, no DC spike, and full open sourced software, try:

   #The mini supports a 6Mhz passband, the v2 supports a 10Mhz passband
   https://v3.airspy.us/product/a-airspy-mini/

--David

vk2auj, 2019-01-30:
So ... I've tried another dongle, same issue, perhaps more resilient but still doing this.  A friend of mine having similar issue.  It could be because in summer we get low 40C (104F) temperature, 38C in the house down to 28C (82F) min at night.  Might be ok in winter waking up to a cool 12C (53F) in the house!  Anyhow, I've decide to heatsink the dongle, using a long TO3 heatsink cut to the length of the dongle and machined to match the 3 ICs, the regulator, capacitors and other components.  I've used a thermal pad for the under board side.  The thing never gets hot ... however nothing beat the sensitivity of my AOR 3000!
[image]

[image]

dranch, 2019-02-01:
Wow.. that's a pretty massive heat sink but I think you're investing way too much money into cheap, low performance SDRs.  Consider looking at better units like Airspy, etc. and while they might also have issues with high temps, your time investment to cool them off will be more worth it.  :-)   Anyway, I think you've demonstrated that your issue isn't with Direwolf so please close this issue