wb2osz/direwolf #184: [PATCH] Adding channel choice to sending transmit calibration tones -x option

f6bvp, 2018-12-17:
This patch adds channel 0 or 1 choice for sending transmit calibration tones.
Checked with UDRC-II and DRAWS hat cards on RaspBerry Pi 3B+
Compass Linux beta-5 version.

Bernard, f6bvp 

[direwolf_calibration_channel.txt](https://github.com/wb2osz/direwolf/files/2687995/direwolf_calibration_channel.txt)

f6bvp, 2018-12-21:
Improved patch version -x n option sends calibration tones only on available channel.
With ACHANNELS 2 in direwolf.conf,  online -x option may be 0 or 1
With ACHANNELS 1 -x 0 is the only value accepted, ortherwise error message.

Bernard, f6bvp

[direwolf.patch2.txt](https://github.com/wb2osz/direwolf/files/2703361/direwolf.patch2.txt)

wb2osz, 2025-03-30:
I think this was done a few years ago but I forgot to close this.

closed 2025-03-30 by wb2osz