wb2osz/direwolf #185: Txing with a busy channel. Wait for clear frequency

Skinoku, 2018-12-29:
I was using my station with your fine software and it works very well, but I can't figure out how to prevent it from transmitting when channel is busy, creating collision. 
I tried keeping squelch from transceiver open in order to delay TX for educational purposes but it still TX immediately any kind of packet, digipeating or iGate. Is it my fault (I have to change the config file) or there's an issue with DW? I'm running on windows latest stable release. Thanks

Skinoku, 2018-12-30:
I tried with a fresh release of DW, with just audio ports setted up and on multiple PC with different soundcards, nothing fancy.
It immediately pops up those two lines and nothing more also with busy channel. I tried hooking up a normal mike and shouting into it and it still prints those two lines and nothing more.
Of course all the rest works fine, it decodes and TX packet with no problems.
`DCD 0 = 1
DCD 0 = 0`
If I understand it right it should change DCD 1/0 when audio is detected, but it doesn't.
Thanks

dranch, 2018-12-30:
This isn't a bug so this Github issue should be closed and the conversation moved to the Direwolf Yahoo group at direwolf_packet@yahoogroups.com .  Until then, this might help.

Define "channel busy" in *your* situation?  Direwolf is designed to run on a dedicated packet frequency with your radio in open squelch mode where it's always listening.  When it hears a valid packet radio preamble (aka.  HDLC sequence), it changes the DCD (Data Carrier Detect) to "1".  That means the frequency has become busy and it's AX.25 protocol engine is now going to try decoding the incoming packet.  DCD does not represent that any "audio" is present and thus is disabling any possibility of Direwolf from transmitting.  

If you're looking to have Direwolf run alongside other modes on a non-dedicated packet frequency, it can get very difficult.  For example, maybe you're looking to have Direwolf run along side SSB or FM phone (aka spoke words).  There are lots of pauses between spoke words, different speaking cadences, etc. that all can false trigger.  There is the Direwolf TXINH configuration parameter described in the UserGuide that could be used with an external program to mimic what you want.  

This could become very complex to both properly clone the incoming audio to both the program and Direwolf, avoid false positives from blocking Direwolf's TX, etc. as there is a LOT to consider here.  For example, see:

   https://web.cs.wpi.edu/~cs529/s15/projects/proj1/proj1.pptx

Here are some possible simple examples of how to detect silence:

   https://unix.stackexchange.com/questions/384290/detecting-when-only-silence-is-being-played

--David
KI6ZHD

Skinoku, 2019-02-21:
Thanks a lot for help.
I understood what you mean by channel BUSY and it's ok.
The problem was probably that some stations were so over deviating and or weak that couldn't even trigger TX inhibition.
Issue for me is closed.
Cheers

closed 2019-02-21 by Skinoku