wb2osz/direwolf #186: Feature Request: Copy KISS frames betweeen TCP KISS Clients

radar27, 2018-12-31:
The ability to have up to 3 (or more if recompiled) TCP KISS clients is very useful in some installations.  While co-located systems would ideally talk to each other directly (not using a radio), in some instances it is desirable for two or more TCP KISS clients be able to see each other's traffic and even connect to each other.  

A recently deployed Direwolf+BPQ implementation is using pilinbpq to provide Winlink RMS connections.  The site owner also wanted to be able to periodically test the RMS by using Winlink Express and pointing it to the TCP KISS port on Direwolf.  Winlink Express and Direwolf connected up just fine, but the pilinbpq instance does not copy it's frames and as such never responds to a connection request, so the test connections are unable to succeed.

Would it be possible to add some type of "KISS Client Crosslink" option that would send KISS frames both to Direwolf to go out over the air as well as all other TCP KISS systems?

```
ClientA -> KISS Frame -> Direwolf
                 |
                 +----> ClientB
                 |
                 +----> ClientC
```
The transmission ClientA made would go to Direwolf and out over the air, in addition, ClientB and ClientC would also copy the transmission as if it came over the air.  If ClientA were to initiate a request to ClientB, ClientB could then respond using the same TCP KISS connection and allow the two to communicate.

73!

wb2osz, 2019-01-02:
This is now in the "dev" branch.  
Select the dev branch by typing "git checkout dev" before compiling.
Enable the feature by putting KISSCOPY in the configuration file.

Take a look at "kissutil" for troubleshooting KISS issues.
I was able to test this new feature by running 3 instances of kissutil at the same time.
An AX.25 packet (in usual monitor format) typed into one showed up in the other two.

Waiting for feedback before closing.

radar27, 2019-01-02:
I've tested this across several instances at 3 sites and everything works great.

Thanks!

wb2osz, 2019-11-24:
implemented and tested.  Closing issue.

closed 2019-11-24 by wb2osz