wb2osz/direwolf #188: IGate not started on Ubuntu 16.04

handiko, 2019-01-02:
I tried Direwolf release 1.5 (October 2018, but I downloaded it at Dec 2018) on Linux Ubuntu 16.04 machine. The APRS decoding, digipeating works well, but IGate function would not start.

In the windows machine, the Windows (same 1.5) release would start the IGate functionality just fine. 
Same laptop, same network, even same configuration file.

Any idea?

wb2osz, 2019-01-03:
When direwolf starts up, it should display some information about the configuration and progress of connecting to an APRS-IS server.   What do you see on the screen?  How does it differ between the two cases?

It should not be necessary to run direwolf as root (sudo).   It could be a permission issue.  What does the "id" command produce?

This should really go in the discussion group.   This "issues" section is meant for bug reports (defects) and enhancement requests.

handiko, 2019-01-05:
Hi.. a moment ago, I tried again with and without sudo to use it as an IGate, and guess what? Now it works!! Now I wonder what I did or how I did it the last time. As a bonus, now I see a bunch of other packets from APRS-IS on the direwolf screen, labeled as [ig->tx] or some sort...

Btw, before this test, I logged in once from my APRS Droid app on my smartphone directly to the APRS-IS using the same callsign and SSID. Maybe it has an effect?

I think the issue is solved for me by now. Thanks for your hard work.

73, de YD1SDL

handiko, 2019-01-15:
> This should really go in the discussion group. This "issues" section is meant for bug reports (defects) and enhancement requests.

Hi, I want to join the discussion group. Where I can find them?

dranch, 2019-01-15:
direwolf_packet@yahoogroups.com

wb2osz, 2020-03-21:
Discussion group has moved to   https://groups.io/g/direwolf

closed 2020-03-21 by wb2osz