wb2osz/direwolf #192: Failure to start: Could not set number of channels, invalid argument

cjritola, 2019-01-30:
Latest git. Install from raspbian repos works but build gives error for all sound devices.

> $ direwolf -p
Dire Wolf version 1.5
Includes optional support for:  cm108-ptt
Reading config file direwolf.conf
Audio device for both receive and transmit: hw:2,0  (channel 0)
Could not set number of audio channels.
Invalid argument
for hw:2,0 output.
Pointless to continue without audio device.

> $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Camera [USB 2.0 Camera], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Creation [Cable Creation], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

>  $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Creation [Cable Creation], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

> $ cat direwolf.conf
CHANNEL 0
 MYCALL AD8BS-5
 MODEM 1200
 TXDELAY 40
 TXTAIL 8
 FIX_BITS 1 AX25
 AGWPORT 8001
 KISSPORT 0
 #FULLDUP ON
 ADEVICE hw:2,0
 #ARATE 44100
 #ACHANNELS 2

> $ uname -a
Linux banana 4.9.70-v7+ #1068 SMP Mon Dec 18 22:12:55 GMT 2017 armv7l GNU/Linux

wb2osz, 2019-01-30:
The documentation says that you should use plughw:2,0 rather than hw:2,0.  Does that help?

cjritola, 2019-01-31:
That did the trick, thanks. Guess I missed that part. 

Close/Invalid.

closed 2019-01-31 by cjritola