wb2osz/direwolf #193: gpsd freezes when using Direwolf on Raspberry Pi -- Raspbian Jesse

petitnj, 2019-02-04:
Run GPSD on Raspberry Pi to pass time and location to Direwolf 1.5. Starts up and runs for a few hours but then "GPS timeout" occurs. Direwolf continues but the time is not updated. Using a USB based GPS receiver from Adafruit. NTP finds the GPS input just fine and updates for a while. No indication on any of the direwolf or system logs. 

The only way to correct is a reboot of the Raspberry Pi. 

Anyone have any experience with this?

dranch, 2019-02-05:
Hello Petitnj,
It's unclear what your ask is here but this doesn't sound like a bug per-se as there haven't been any other reports of GPS issues with Direwolf.  To get better support, please close this issue and create a new thread on direwolf_packet@yahoogroups.com.  In that new thread, maybe you can answer some of the following questions to get you going:

Do you feel that Direwolf is crashing gpsd?  Maybe you can shutdown Direwolf and just run "gpstl" or "cgps" for several hours and see if your system continues to display GPS data.  If it does, I would then recommend to try again with Direwolf but leave your radio *off*.  The inability for the system to TX would isolate if your data transmissions are creating any RFI issues into the GPS and/or your Raspberry Pi.

wb2osz, 2019-02-06:
I did a search for gpsd timeout raspberry pi, and found a couple related items:

https://raspberrypi.stackexchange.com/questions/30183/gpsd-data-receiving-issues/67767
https://www.raspberrypi.org/forums/viewtopic.php?p=712458

Can you provide more details?  What is the exact GPS model?  I don't see a USB GPS from Adafruit.     What does gpsd configuration look like?  What does direwolf config file look like?  Does the same thing happen when direwolf is not involved?  You already mentioned that the system log files did not contain anything interesting.

It might be better to continue with email rather than here due to the size of information.   

wb2osz  \\  comcast \\ net

petitnj, 2019-02-06:
Using the Adafruit GPS ultimate breakout connected to the RPi with the GPS external antenna, 
RP-SMA to uFL/u.FL/IPX/IPEX RF Adapter Cable, and the USB to Serial Cable. This GPS system has proven years of success in our magnetometer systems for PC timing. Additionally the direwolf.conf has the standard lines for GPSD

GPSD
TBEACON every=30 symbo="digi" power=50 height=20 gain=4 comment="Harris, MN" WIDE1-1,WIDE2-1
SMARTBEACON 60 2:00 5 10:00 0:45 30 255

I entered this thread to see if anyone else had the problem. With the recent library updates (Feb 5 2019) of the operating system, the freeze up of gpsd has disappeared (knock on wood). I am closing this thread for now and will reopen if it reappears.

closed 2019-02-06 by petitnj