wb2osz/direwolf #196: Compile error against gpsd-devel 3.18.1-2.fc30

matt-domsch-sp, 2019-02-11:
Pre-release version of Fedora 30, Fedora Release Engineering did a mass rebuild, using newer GCC version 9.0.1-0.4.fc30.  Direwolf fails to build, complaining of a version mismatch with libgps, in use is gpsd-devel   version 3.18.1-2.fc30.

gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD -DUSE_HAMLIB -DUSE_CM108   -c -o dwgpsd.o dwgpsd.c
BUILDSTDERR: dwgpsd.c:61:2: error: #error libgps API version might be incompatible.
BUILDSTDERR:    61 | #error libgps API version might be incompatible.
BUILDSTDERR:       |  ^~~~~
BUILDSTDERR: dwgpsd.c: In function 'read_gpsd_thread':
BUILDSTDERR: dwgpsd.c:260:8: error: too few arguments to function 'gps_read'
BUILDSTDERR:   260 |    if (gps_read (&gpsdata) == -1) {
BUILDSTDERR:       |        ^~~~~~~~
BUILDSTDERR: In file included from dwgpsd.c:55:
BUILDSTDERR: /usr/include/gps.h:2122:12: note: declared here
BUILDSTDERR:  2122 | extern int gps_read(struct gps_data_t *, char *message, int message_len);
BUILDSTDERR:       |            ^~~~~~~~
BUILDSTDERR: make: *** [<builtin>: dwgpsd.o] Error 1

[build.log.txt](https://github.com/wb2osz/direwolf/files/2852481/build.log.txt)
[root.log.txt](https://github.com/wb2osz/direwolf/files/2852489/root.log.txt)

mdomsch, 2019-02-13:
Fedora bug https://bugzilla.redhat.com/show_bug.cgi?id=1674812 first reported it.

wb2osz, 2019-02-14:
It appears very easy to fix.
Need to do some live testing on both ARM and x86_64 before releasing the change.

wb2osz, 2019-02-18:
Fixed in git commit  f1b54df64766bf9f2cb95f71038a9d9c9579b172

mdomsch, 2019-02-18:
Thank you.  I've pushed 1.5 plus this patch into Fedora 28 (testing), 29 (testing), and 30 (pre-release).  It should appear in the updates-testing repositories for F28 and F29, and in the main repository for F30, shortly.

closed 2019-02-19 by matt-domsch-sp