wb2osz/direwolf #198: RX-only IGate with Nooelec SmarTee on RPi 3+ B Rev 1.2 stops receiving randomly

TheButterZone, 2019-02-26:
$ sudo systemctl status direwolf
● direwolf.service - Direwolf service
   Loaded: loaded (/etc/systemd/system/direwolf.service; enabled; vendor preset:
   Active: active (running) since Tue 2019-02-26 00:00:05 PST; 11h ago
 Main PID: 315 (dw.sh)
   CGroup: /system.slice/direwolf.service
           ├─315 /bin/sh /usr/local/bin/dw.sh
           ├─319 rtl_fm -f 144.39M -
           └─320 direwolf -c /home/pi/sdr.conf -r 24000 -D 1 -

Feb 26 04:57:03 pi dw.sh[315]: [ig>tx] WI6RE-8>APT311,KELLER,WIDE1,qAR,W6HRO:>70
Feb 26 04:57:06 pi dw.sh[315]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only 
Feb 26 05:16:10 pi dw.sh[315]: 
Feb 26 05:16:10 pi dw.sh[315]: [ig>tx] KQ6NO-13>APRS,WA6ZSN,WIDE1*,WIDE2-1,qAO,W
Feb 26 05:56:06 pi dw.sh[315]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only 
Feb 26 06:55:06 pi dw.sh[315]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only 
Feb 26 07:54:06 pi dw.sh[315]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only 
Feb 26 08:53:06 pi dw.sh[315]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only 
Feb 26 09:52:06 pi dw.sh[315]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only 
Feb 26 10:51:06 pi dw.sh[315]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only

TheButterZone, 2019-02-26:
Just checked status again at 14:22 local:

Feb 26 13:45:38 pi dw.sh[1219]: [ig>tx] W6MKO-9>DDFTUP,WIDE1-1,WIDE2-1,qAR,N6LXX-10:`.\IlA@k/`"4+}_%
Feb 26 13:45:39 pi dw.sh[1219]: 
Feb 26 13:45:39 pi dw.sh[1219]: [ig>tx] N6EX-4>APNU19,qAR,WA6IAJ-1:!3336.35N111748.63W#PHG5660/W1 for Orange County
Feb 26 13:45:43 pi dw.sh[1219]: 
Feb 26 13:45:43 pi dw.sh[1219]: [ig>tx] XE2GF-10>APMI06,TCPIP*,qAC,T2MEXICO:@262145z3230.09N/11702.29WnPHG56604 Tijuana BC xe2gf@
Feb 26 13:45:54 pi dw.sh[1219]: [ig] N6ATF-10>APDW15:!3245.  NR11706.  W&RX-only IGate | DireWolf on RPi+RTL-SDR
Feb 26 13:46:11 pi dw.sh[1219]: 
Feb 26 13:46:11 pi dw.sh[1219]: [ig>tx] KQ6NO-13>APRS,WA6ZSN,WIDE1*,WIDE2-1,qAO,W5AFK-10:@041223z3421.53N/11905.22W_000/000g004t0
Feb 26 13:49:04 pi dw.sh[1219]: 
Feb 26 13:49:04 pi dw.sh[1219]: [ig>tx] WA6IAJ-1>APMI06,TCPIP*,qAC,T2BC:@262149z3344.19N/11802.17W#Huntington Beach, CA iGate/Dig

wb2osz, 2019-02-27:
Is the SDR very hot?  There was another recent case where the SDR would overheat and shutdown.
Based on the information provided so far, I see no reason to believe that this is a bug in the direwolf application so this should not be in the "issues" section.  It would be more appropriate for the discussion group.   https://groups.yahoo.com/neo/groups/direwolf_packet/info;

closed 2019-03-06 by TheButterZone