wb2osz/direwolf #203: TBEACON frames not copies to TCP KISS

AlyBadawy, 2019-03-31:
Not sure if this is a bug or was done intentionally, but the beacons generated by the Direwolf configuration file, are not copies on the TNC KISS port. all other beacons (from TCP KISS) are sent to audio, and all other received on audio are copied to the TCP KISS port. 

The reason why I am asking, is I am using another software (APRX) as the core of the aprs engine, and direwolf as a KISS TNC. however, I need to beacon the location based on GPS from direwolf. this happens on the audio side, out to radio and then to RF. I am still interested in i-gating that beacon via the internet through APRX. 

Ideas?

wb2osz, 2019-03-31:
Look in the User Guide, Beaconing section, "SENDTO" option.
You can use "R" and a number to simulate reception from the specified channel.
This is sent to the attached applications (KISS or AGW network interface) as if they were received over the radio.

AlyBadawy, 2019-04-01:
Perfect. Exactly what I needed. Thanks a lot.

closed 2019-04-01 by AlyBadawy