wb2osz/direwolf #206: PBEACON gone wild

craigerl, 2019-04-04:
Direwolf 1.5 was issuing a beacon every second, for a minute or so (I killed the power to the Tx after a dozen).   It was running for a few days w/o problems.  During this test, the only difference was a disconnected internet.   Any reason it would send out PBEACONS as fast as possible?   Raspbian, pi-zero, thus compiled from a git clone last week.   This is a "personal" digi only interested in traffic to/from my callsign -- conf file below.

Digipeater WIDE2 (probably VACA) audio level = 52(27/22)   [NONE]   _|||||___
[0.3] KE5HXX-2>S7RUPU,W6CX-3,VACA,WIDE2*:`2'Ko?W>/`"4C}_%<0x0d>

RANCHO audio level = 58(27/14)   [NONE]   _||||||||
[0.4] RANCHO>APN391:!3843.52N112038.40W#PHG2560/NCAn,TEMPn/AG6AU/Rancho del Sol,Camino,CA/A=003400<0x0d>

Digipeater K6FGA-1 audio level = 49(25/16)   [NONE]   _|||||||:
[0.4] KK7AIR-9>S8UTRU,K6FGA-1*:`1\Xl#/j/`"7y}_%<0x0d>

Digipeater GTOWN audio level = 51(26/16)   [NONE]   _|||||||_
[0.4] KK7AIR-9>S8UTRU,GTOWN*:`1\Xl#/j/`"7y}_%<0x0d>

Digipeater WIDE2 (probably VACA) audio level = 53(24/22)   [SINGLE]   ___::____
[0.3] KF6GPE>APWW10,WR6ABD,VACA,WIDE2*:@212854h3710.60N/12208.80W-@kf6gpe / kf6gpe.org

Digipeater K6FGA-1 audio level = 50(27/16)   [NONE]   _||||||__
[0.3] N6YP-11>SWRWYS,W6CX-3,K6FGA-1*,WIDE2:`2)Jn>%>/`"4%}_%<0x0d>

Digipeater WIDE1 (probably VACA) audio level = 53(25/21)   [NONE]   ||||||___
[0.2] K6PKL-2>APOTW1,VACA,WIDE1*:!3748.51N/12112.44W_180/002g003t067V136P000h73b10211OTW1

Digipeater WIDE2 (probably VACA) audio level = 51(26/21)   [NONE]   _||||||__
[0.3] KE6FOA-3>APAGW,W6BXN-3,VACA,WIDE2*:=3724.39N/12050.68W_PHG5140AGWTRACKER

Digipeater K6FGA-1 audio level = 47(24/16)   [NONE]   _|||||||_
[0.4] KE5HXX-2>S7RTVV,W6CX-3,K6FGA-1*,WIDE2:`2'^m5u>/`"4B}_%<0x0d>
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[... 39 lines cut ...]
#PBEACON  sendto=IG compress=1 delay=00:30 every=20:00 symbol="igate" overlay=R lat=38^54.70N long=120^56.15W comment="Limited local digi, only specific callsigns on RF, part time"
PBEACON            compress=1 delay=00:12 every=20:00 symbol="igate" overlay=R lat=38^54.70N long=120^56.15W via=WIDE1-1 power=5 height=5 gain=3 comment="Limited local digi, only specific callsigns on RF, part time"

# Radio->Radio  Only traffic from me
FILTER 0  0 b/KM6LYW*

# IG->Radio     Only traffic to me
FILTER IG 0 g/KM6LYW*

# Radio->IG     Only traffic from me
FILTER 0 IG b/KM6LYW*

# Tell server to only send me message type packets with me as addressee
IGFILTER  g/KM6LYW*

DCD GPIO 25
PTT GPIO 24
DWAIT 0
TXDELAY 30
TXTAIL 10
IGTXVIA 0 WIDE1-1
DIGIPEAT 0 0 ^WIDE[3-7]-[1-7]$ ^WIDE[12]-[12]$
AGWPORT 8000
KISSPORT 8001
ADEVICE plughw:0,0

craigerl, 2019-04-05:
following up after going through the code.  According to beacon.c, if the system clocks shifts forward, direwolf will emit all the pbeacons it missed during that interval all at once, until it's caught up.   For example, if the internet is unavailable during boot, ntp cannot update the system time.  Direwolf then starts, the internet becomes available, and ntp shifts the clock foward three hours.  If direwolf is configured to send a pbeacon every 20 minutes, it will quickly send *nine* consecutive pbeacons.

In beacon.c below, just below the comment "Don't take relative to 'now' in case there was some delay, is where the bug exists.  It should, in fact, be relative to "now" in case the system clock shifted forward... or at least be a configuration option.

```

/*
 * Send if the time has arrived.
 */
	  for (j=0; j<g_misc_config_p->num_beacons; j++) {

	    struct beacon_s *bp = & (g_misc_config_p->beacon[j]);

	    if (bp->btype == BEACON_IGNORE)
	      continue;

	    **if (bp->next <= now) {**

	      /* Send the beacon. */

	      beacon_send (j, &gpsinfo);

	    .
.
.
.
.
.

	      else {
	        /* Non-tracker beacon, fixed spacing. */
		/* Increment by 'every' so slotted times come out right. */
	        **/* i.e. Don't take relative to now in case there was some delay. */**

	        **bp->next += bp->every;**
	      }

	    }  /* if time to send it */

	  }  /* for each configured beacon */

	}  /* do forever */

```

craigerl, 2019-04-05:
Confirmed bug by setting clock ahead 3 hours after direwolf had started.  With a 20minute pbeacon configured, I was able to replicate the problem.

```

Digipeater GTOWN audio level = 51(26/17)   [NONE]   _|_||||||
[0.5] N6QGS-8>SX3XTQ,GTOWN*,WIDE2-1:`1)blIr-/]"4H}=<0x0d>

Digipeater K6FGA-1 audio level = 45(23/15)   [NONE]   _:||||||:
[0.4] N6QGS-8>SX3XTQ,GTOWN,K6FGA-1*,WIDE2:`1)blIr-/]"4H}=<0x0d>
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time
[0L] KM6LYW-1>APDW15,WIDE1-1:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time

Digipeater GTOWN audio level = 51(25/16)   [NONE]   _|||||___
[0.3] KM6LYW-1>APDW15,GTOWN*:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time

Digipeater DANAHR audio level = 85(44/16)   [NONE]   ___||||||
[0.5] N6QGS-8>SX3XTQ,DANAHR*,WIDE2-1:`1)blIr-/]"4H}=<0x0d>

Digipeater DANAHR audio level = 85(42/16)   [NONE]   ___||||||
[0.5] KM6LYW-1>APDW15,DANAHR*:!R:l&f/uL<&{&GLimited local digi, only specific callsigns on RF, part time

```

craigerl, 2019-04-05:
Proposed patch.  Tested by moving clock forward, and back....

```

              else {
                /* Non-tracker beacon, fixed spacing. */
                /* Increment by 'every' so slotted times come out right. */
                /* i.e. Don't take relative to now in case there was some delay. */

                bp->next += bp->every;

                /* craiger: if next beacon is scheduled in the past, then set next beacon relative to now (happens when NTP pushes clock AHEAD) */
                /* fixme: if NTP sets clock BACK an hour, this thread will sleep for that hour */
                if ( bp->next < now ) {
                    bp->next = now + bp->every;
                    dw_printf("\nSystem clock appears to have jumped forward.  Beacon schedule updated.\n\n");
                }

              }

            }  /* if time to send it */

```

juchong, 2019-05-31:
@craigerl I would recommend forking the repository and submitting a pull request. It's the cleanest way to contribute to the project since you've already pinpointed the source of the issue and implemented a solution.

craigerl, 2020-12-02:
Pull request #301 created

sardylan, 2021-04-09:
You have, perhaps, just resolved the issue (I'm going to test your PR asap), but I add my similar problem.

I'm working on a replacement for an hardware Digipeater using a RPi 3 with Direwolf, no Internet, and a GPS + PPS system for clock sync. Everything will be installed in the cabin on top of the mountain, with an hour plus of off-road driving for any debug or problem solving, so I'm testing all the systems in my station at home.
I'm keeping GPS receiver inside, so satellite signal is not very good. In this way I'm able to reproduce a similar problem, perhaps due to clock sync.

Every time I switch off and on the RPi, the system clock restart with a wrong datetime. GPS needs some time to be back in sync with satellites, so Direwolf process starts with bad clock. After few moments the GPS acquires a good signal, and ntpd process suddenly changes the system date and time based on GPS + PPS informations.
When the first beacon transmission triggers, the PTT remains always ON and the log is flooded with:

```
Apr  9 21:45:10 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:10 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:10 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:10 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:10 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:10 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
Apr  9 21:45:11 raspberrypi direwolf[667]: Transmit packet queue for channel 0 is too long.  Discarding packet.
Apr  9 21:45:11 raspberrypi direwolf[667]: Perhaps the channel is so busy there is no opportunity to send.
```

As you can see, several messages per seconds are added in logfile.
The only way to let everything working correctly is to restart Direwolf process after clock sync.

Since our Digipeater is powered with batteries and solar cells, the problem could occur often, especially in periods, like in the winter, in which the daylight is not enough for a full battery charge.

I will let you know if your patch resolve my problem.

craigerl, 2021-07-06:
I haven't seen that in the log in my examples.  did my patch fix it in the long run?   it was easy to replicate, just update the clock after direwolf starts, and it transmits full time with pbeacon after pbeacon.

wb2osz, 2022-02-16:
I put your patch into the 'dev' development branch.
I was sitting on this because it will break the time slot method.
Leaving the case open as a reminder to go back and fix the time slot case.

craigerl, 2022-02-16:
Thanks, this issue is further exacerbated by Pi's getting time from a GPS, which often happens after direwolf starts.  If a Pi was off for three months, or has a read-only filesystem with no cached time, for example, direwolf will send three months worth of pbeacons jamming the frequency for potentially hours.  If time changes, the slot method is broken regardless? afaict?

craigerl, 2022-05-28:
additionally, in Winlink/ax.25 mode, direwolf jams the channel with QST UI packets after ntp sets the clock on a Pi,

```
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
[0L] KM6LYW>QST:(UI cc=00, p/f=0)<0x00><0x03><0x00><0xcc><0x07><0x04><0x00><0x01><0x96><0x9a>l<0x98><0xb2><0xae><0x00>,8<0x04><0xdf><0x00><0x00><0x00><0x00><0x00><0x00><0x00>,<0xbe>({
```

wb2osz, 2022-05-28:
I should probably be using clock_gettime (CLOCK_MONOTONIC) so the timing will be independent of the real time clock.

sq2mo, 2023-07-11:
Issue still in place for 1.7-dev-A on Raspberry Pi.
Direwolf goes brrrrr (CBEACON in my case) once NTP is synced.

dranch, 2023-07-11:
This is evidently not simple to fix properly within Direwolf.  A very effective and inexpensive workaround is to connect up a $7 DS3231 real time clock (RTC).  This also has other benefits such as improved bootup log timestamping, a built in temperature sensor, etc.  Here is a guide I wrote up or that for a Raspberry Pi:
https://www.trinityos.com/HAM/CentosDigitalModes/RPi/rpi4-setup.html#60b.appendix-i2c-rtc

wb2osz, 2023-07-12:
According to the comments here, and examination of the source code revision history, a work-around was added 15 Feb 2022.
This would correspond to 1.7 DEV version D.
Try the latest on the dev branch and try again.

sq2mo, 2023-07-12:
@wb2osz thanks for your comment.

Ran a quick test stopping the ntpd service and setting the date to 2000 or so, then launched direwolf with cbeacon configured. Reenabled ntpd and... no more hundreds of beacons, just this message:

`System clock appears to have jumped forward.  Beacon schedule updated.`

So it seems the issue is solved. Do you plan to release a dev snapshot in nearest future?