wb2osz/direwolf #208: APRS Station Symbol Change

MrCre8ive, 2019-04-19:
In The .Conf file for Direwolf, there is a option to change the symbol that's shown on the map. By default is set for "digi". I want it to be the "house" symbol. There is no documentation on how to change the symbol to a certain icon. Is there a chart of icons to words for this? What do i put in between the quotes to change my icon to the house icon?

(Sorry if im using the wrong terms, this is only my 2nd Raspberry PI Project! Comment if you have questions about what im trying to ask)

dranch, 2019-04-19:
I suppose the documentation can be enhanced a bit here but..

   - Official (non-graphical) symbol list: http://www.aprs.org/symbols/symbols-new.txt
   - Official (non-graphical) SSID list: http://aprs.org/aprs11/SSIDs.txt

   - A good graphical table: https://www.aprsdirect.com/symbol/list

   - Another good graphical table: https://www.yachttrack.org/info_camper/downloads/APRS_Symbol_Chart.pdf

   - Another good graphical table: http://aprs.od5.org/symbols.htm

MrCre8ive, 2019-04-20:
this table will work with direwolf? [http://aprs.od5.org/symbols.htm](url)
Its the last table you listed.
If I put "-" in the quotes, I should get the house symbol, correct?

dranch, 2019-04-20:
Close.  When configuring a symbol, you need to first select the table and then the specific symbol in that table.  Here is what I use for an APRS Object with a house symbol:

PBEACON delay=00:15 every=60 SYMBOL=/- lat=37.343595 lon=-121.999800 power=4 height=20 gain=6 comment=" RaspberryPi3 + 5W Direwolf 1.5r TNC/Igate + Linpac chat in Santa Clara" via=WIDE1-1

Anyway.. using a Github "issue" is not for support but I responded as we can use this ticket for WB2OSZ to enhance the documentation with some symbol URLs.  if you have any other questions, please subscribe and post to the Direwolf email list at direwolf_packet@yahoogroups.com

--David
KI6ZHD

wb2osz, 2019-11-24:
Run direwolf with the "-S"  (dash, upper case S) option to see a list of all the symbols.

Section 9.9, Beaconing, in the User Guide, discusses symbols.  Maybe this section needs to be rewritten.  What part is not clear?

kh6bfd, 2021-11-28:
I hope this is the right place to ask.  I'm trying to set up my third Digipeater for another repeater location on the south side of my island.  I was having trouble with keeping the Digipeater running and now it runs.  But, it doesn't show up on the map.  As far as I can tell I did everything the same in the script as mother digipeaters.  But, nothing.  Also, when I beacon, it doesn't repeat back my location like the other digipeaters.  Something new in the newer version?  I believe it is 1.6 that I am working with and the other two are 1.5

dranch, 2021-11-29:
KH6BFD: This is not the right place to get support as this area is used for real bugs or feature enhancements.  Your issue also does not have anything to do with this specific Github issue.  Please join and post your support question to direwolf@groups.io