wb2osz/direwolf #210: CM108 driver data and mask possibly reversed

pfiliberti, 2019-05-28:
I believe that possibly the data and mask sent in function cm108_write may be reversed.
I had problems working with my device and i looked at some other code for diagnosing
uri's found here: https://github.com/AllStarLink/uridiag. I have mine now set to the followin:
io[2]=iodata;
io[3]=iomask; 

That corrected the problem for me. Also my device shows it has a product id of 0x013c which direwolf has as CMEDIA_PID_HS100 however the diag code has it as CMEDIA_PID_CM108AH.
My device was not being recognized until I changed that so maybe we need to revisit the vendor ids? Thanks all hope this helps. 73

wb2osz, 2019-05-29:
Thanks for pointing that out.  After reviewing the CMedia datasheets, we see that data (high, low) comes before the mode (input, output).

The product ID is problematic.  It is sometimes listed as a range in the datasheets and is sometimes programmable.  This is what I found in the datasheets.

CM108: 0008 - 000f
CM119: 0008 - 000f - programmable by MSEL and MODE pin.
CM109: 0008 - 000f - programmable by MSEL and MODE pin.
CM108B: 0012
CM119B: 0013
CM108AH: 0139 - product id programmable by MSEL and MODE pin.
CM119A: 013a
HS100: 013c

To make things more confusing,  <QUOTE>

"PID changes with the jumper pin setting so different setting have different PID. For customized product, customer can attach a 93C46 EEPROM to override the embedded VID, PID and provide addition serial number for each set. CM119A automatically detects 93C46 existence and performs the overwrite function during power up."

wb2osz, 2019-07-06:
Fixed in dev branch and will be in release 1.6. 

342989b10017b58555477bba57afb9470c3337df

closed 2019-07-06 by wb2osz