wb2osz/direwolf #212: Part of Direwolf IGate stops working.

KW0O, 2019-06-15:
Set up as IGate on Pi 3 Stretch, it passes packets up to internet, but after a few hours to a week it stops and APRS.fi graph shows no activity. However screen shows ap>tx (light green) still updating. I can end Direwolf and restart manually but problem will return after random interval. How to fix?

KW0O, 2019-06-15:
Forgot to mention, this is Direwolf 1.5, and position packets continue without interruption even when packets going to T2 server stop.

KW0O, 2019-07-06:
I don't know why this is shown as closed, I didn't close it. The issue still exists and now sometimes the position packets stop for awhile and then restart, while the RF packets are still uploaded to the internet. I need to fix this as KW0O-10 is a major IGate in Cedar Rapids and it a nuisance to keep restarting Direwolf.

How can I fix it?

KW0O, 2019-07-06:
I need to fix this problem of Direwolf mis-behaving.

ab0tj, 2019-07-06:
It was probably closed because this isn't the right place to ask support questions. Try the Yahoo group: https://groups.yahoo.com/neo/groups/direwolf_packet/

wb2osz, 2019-07-06:
Look closely.  it says:  "KW0O closed this 21 days ago".  There are two buttons at the bottom.  One says "close and comment".  The other is only "comment."

You really haven't provided much useful information for troubleshooting this.  The User Guide, section 9.10.7, suggests:

Log Everything Going In and Out

There are many command line options available to customize the information seen for troubleshooting. 

You might want to see everything on the radio, two way communication with the Internet Server, and details on what the filtering is doing in between.  These would be good options to use:
-t 0
Turn off text color which might cause issues later when viewing in a text editor.   (Starting with version 1.6, this is done automatically when stdout is redirected to something other than a terminal.  You can override this behavior  with the “-t” command line option and a non-zero value.)

-q hd
	Suppress the lines with audio level and explanation of what frame means.

-d ii
	Display details on what the IGate function is doing.

-d fff  
	Display details on what packet filtering is doing.

If you are using Linux, capture the output to a file like this:

direwolf -t 0 -q hd -d ii -d fff  |  tee aprs,log

The tee command displays incoming text real time and also saves it to a file for later review.

Take a look at this for more information about how the IGate works and troubleshooting tips.

https://github.com/wb2osz/direwolf/raw/dev/doc/Successful-APRS-IGate-Operation.pdf

wb2osz, 2019-11-24:
Reopen when more details are available.
The "-d iii" option will provide more information about what is happening.
You also need to be aware that the APRS-IS servers remove duplicates.
If some other IGate already sent the same packet yours will be discarded.

closed 2019-11-24 by wb2osz