wb2osz/direwolf #213: Packets chopped to almost nothing

TheButterZone, 2019-06-21:
Now Raspbian GNU/Linux 9 (stretch) isn’t playing Direwolf 1.5’s packets in their entirety, only a split second of flat tone, through the RPi 3B+ internal soundcard. I plugged in earbuds to confirm this instead of just listening for the packets through another radio, like I successfully did when I set it up.

dranch, 2019-06-21:
It's not clear what your issue is here.  Are you saying that you previously had a fully functioning Direwolf v1.5 setup but after a Raspbian update, it no longer works?  Since you mention you never used a USB adapter, I assume you were only transmitting packets and not receiving them (you need a microphone for that and Raspberry Pis don't have microphone inputs).  If that's all correct so far, can you state what kernel version you're using?  If you use Direwolf's calibrate feature, does that work as expected?

TheButterZone, 2019-06-21:
I don't know when exactly it stopped working. I knew it worked when I initially enabled PBEACON on channel 0. It's still receiving only via RTL-SDR, but that's irrelevant.

4.19.42-v7+

What is "Direwolf's calibrate feature" for audio out?

dranch, 2019-06-21:
When you say you "sent messages back and forth", do you mean over the Internet to the IGATE?  If so, that won't have anything to do with Direwolf's modem side of things.  I see your time-domain capture of the audio but would it be possible for you to attach an mp3 or wav of this audio to this ticket?  Finally, it seems that your using an SDR for listening and the RPi built-in soundcard as output.   Have you rebooted your Rpi to make sure things are in a pristine state.  

Finally, please see how to use the calibrate feature in the Direwolf User-Guide.

TheButterZone, 2019-06-21:
Rebooted more times than I can remember, every single time any technical issue has ever come up throughout my life. I didn't just get into computers yesterday.

[Archive.zip](https://github.com/wb2osz/direwolf/files/3315755/Archive.zip)

dranch, 2019-06-21:
What is your ADEVICE setting?  I'm curious what playback sound system you're using (OSS, Alsa, PulseAudio, etc).

TheButterZone, 2019-06-21:
ADEVICE - plughw:0,0

dranch, 2019-06-21:
Hmmm.. that looks fine.  I also listened to your audio files and the beacon sounds initially ok but obviously too short.  I just made some connected-mode connections to various APRS digipeaters here in the Bay Area, CA on 144.390 so that proves to me that I'm not having having similar issues on my  Raspbian Stretch with Direwolf v1.5 using a recommended Syba USB soundcard.  

The only other thing I could initially say is if you could try using an external USB sound card for the TX audio?   It might be different and it would help narrow things down (generally, I don't think most people use the on-board audio at all).

--David

TheButterZone, 2019-06-22:
I'm not going to spend money to suffer [a known bug](https://github.com/wb2osz/rpi-usb-audio-bug).

petem001, 2019-06-22:
Sorry but most usb sound card have stereo output and a mike in, so if you have a usb sound card you already got all you need.

TheButterZone, 2019-06-22:
> **There is no USB audio adapter. There has never been any USB audio adapter.**

petem001, 2019-06-22:
https://www.amazon.com/Channel-External-Sound-Audio-Adapter/dp/B07796JZ1Q/

At 4.99$ this aint a big invertment

TheButterZone, 2019-06-22:
**> I'm not going to spend money to suffer [a known bug](https://github.com/wb2osz/rpi-usb-audio-bug).**

petem001, 2019-06-22:
Strangely the Allstarlink distribution on the RPI use that kind of usb sound card ( cm108 chips) and the timing is perfect..

This system runs on an old version of Asterisk (the PBX system). A lot more heavy software to run then direwolf.

dranch, 2019-06-22:
Yes, it's been reported that Allstar doesn't have the USB sound card + ALSA TX delay issue that's been  observed  by some with Direwolf.  Viable work arounds are available including the OSS driver instead or using an I2S-based soundcards like the Fe-Pi.  The next step is up to you as alternative ideas have been provided for you to try.

--David

TheButterZone, 2019-06-22:
alsa = 1
to
# alsa = 1

with a save & reboot, made no difference. The Morse beacon continues to be the only audio that plays properly. 

The only idea that isn't the definition of insanity is looking like a full reset & reinstall.

TheButterZone, 2019-07-09:
Reflashed with [Raspbian Buster Lite](https://www.raspberrypi.org/downloads/raspbian/), installed everything over again from scratch, following Direwolf user guide, AX.25 audio out packets again chopped to almost nothing.

dranch, 2019-07-10:
It sounds like your issue revolves around using the Raspberry PI's on-board output only sound card.  You mentioned above that you've tried this with the ALSA driver and had issues.  There is one alternative idea to use the OSS wrapper which seems to help on the delay issue with USB-based sound devices but might not help on I2S based (Fe-Pi sound HAT).  I don't know if this approach would help on the built-in sound device but it might be worth trying since you don't want to add a recommended USB sound device per the Direwolf UserGuide.  The OSS work arounds have been mentioned on the Direwolf Yahoo group so please search the archives there and give that a try.

TheButterZone, 2019-07-10:
[The Direwolf UserGuide should no longer recommend anything that Direwolf's own developer found a bug for.](https://github.com/wb2osz/rpi-usb-audio-bug)

[image]

I can't login to Yahoo.

nayrnet, 2019-07-18:
I'm seeing something similar, brought my station back online after being down for nearly a year and updated it.. I had no problems hearing others but nobody could hear me.. did some volume calibrations and everything coming out sounded like garbage compared to other aprs packets.. Device is oDroid C2 w/Dev Branch on a SignaLink.. 
Moved it over to an x86_64 VM and all is well, people hear me just fine with same radio, same config, same sound card.