wb2osz/direwolf #215: Flashing console output on Raspberry Pi

tlrobinson, 2019-07-14:
[image]

What's the purpose of the white background anyway? Why not respect the user's choice of terminal colors?

wb2osz, 2019-07-14:
Terminal behavior changed on Buster.  Try using the "dev" branch.

Cale, 2019-07-17:
I can confirm the same behavior on the latest Raspbian and latest release version of Direwolf. Interestingly, this also occurred over an SSH session, logged into the Raspberry Pi via Ubuntu.

EtoTen, 2019-07-25:
I just ran into this also after upgrading to Buster. I just deleted master and pulled the dev branch.  The issue is fixed in the dev branch, thank you!

For those who are not up on their Linux CLI... here are the commands you need to do:

```
pi@raspberrypi:~ $ rm -rf direwolf
pi@raspberrypi:~ $ git clone https://github.com/wb2osz/direwolf.git
Cloning into 'direwolf'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 2038 (delta 0), reused 1 (delta 0), pack-reused 2032
Receiving objects: 100% (2038/2038), 122.30 MiB | 2.79 MiB/s, done.
Resolving deltas: 100% (1412/1412), done.
pi@raspberrypi:~ $ cd direwolf
pi@raspberrypi:~/direwolf $ git checkout dev
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'
pi@raspberrypi:~ $ make
....
pi@raspberrypi:~ $ sudo make install

```

dranch, 2019-07-25:
tlrobinson:   If you agree your issue is resolved, please close this ticket

EtoTen, 2019-07-27:
On another note... I can't seem to capture packets on a CM119 and Buster... It appeared to have worked with Stretch. I tried a clean install even and tried both 1.5 and dev branch. Not sure how I can create an issue around it though as everything appears to work and there is no error. Just no packets seem to get decoded at all. Not sure if it is something that I am doing wrong or if its an incompatibility with Buster.

dranch, 2019-07-27:
Please don't hijack someone else's ticket.  This ticket has to deal with blinking text.  Anyway, Direwolf works fine on Raspbian Buster.  If you have issues, please post a question to the direwolf_packet@yahoogroups.com

EtoTen, 2019-07-27:
Wasn't trying to hijack... was just making a note that I am experiencing a related issue as it appears to only be exhibited in Buster for me. Thanks for the suggestion to use the yahoo group. I will do more digging to see if there is a root cause that I can identify.

wb2osz, 2019-07-28:
Did you read Raspberry-Pi-APRS.pdf, section 8, Troubleshooting?

closed 2019-07-28 by wb2osz

Cale, 2020-02-16:
Is the fix for the flashing console expected to make its way from dev to master? Still seeing this issue on the latest master.

dranch, 2020-02-17:
Yes, DEV will make it to master once it's deemed stable and ready for a release.  I would argue that the current DEV is pretty stable already.

Cale, 2020-02-18:
Great, thanks for the insight!

m0nde, 2020-03-12:
I have flashing console. I followed advice by EtoTen. All worked fine up until command pi@raspberrypi:~ $ make
Error was something like nothing to make, cleared screen now. I see files CMakeLists.txt, 
Advice welcome
Nigel M0NDE

dranch, 2020-03-12:
This issue tracker is for bugs.  If you have problems with compiling Direwolf, please subscribe and post your question at direwolf@groups.io .  When you do post your question, please post a specific error or the helpful people there won't be able to help you.

jharwinbarrozo, 2020-04-05:
> I just ran into this also after upgrading to Buster. I just deleted master and pulled the dev branch. The issue is fixed in the dev branch, thank you!
>
> For those who are not up on their Linux CLI... [...]

This won't work if you checkout the branch. DEv branch requires these steps to compile:

cd ~
git clone https://www.github.com/wb2osz/direwolf
cd direwolf
git checkout dev
mkdir build && cd build
cmake ..
make -j4
sudo make install
make install-conf

simontheu, 2020-04-20:
Tried this and no joy with the dev branch. Still have a flashing console, everything built and installed without error.