wb2osz/direwolf #218: Direwolf USB - tty error (Raspberry Pi and IC-7100)

m0zzm, 2019-08-16:
Hello Everyone,

I wonder if you can help me configuring DireWolf on my Raspberry Pi. The hardware I am using are a Raspberry Pi 3 with Rasbian Buster, directly connected to an IC-7100 via a USB cable. I successfully connected Fldigi to the IC-7100 so I have no reasons to believe any of the hardware is faulty.

The issue I am facing with Direwolf when I start the system is the following:

------------------------------------
Dire Wolf version 1.5
Includes optional support for:  cm108-ptt

Reading config file direwolf.conf
Audio input device for receive: stdin  (channel 0)
Audio out device for transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
Converted COM6 device 'PTT' to Linux equivalent '/dev/ttyS5'
ERROR can't open device /dev/ttyS5 for channel 0 PTT control.
No such file or directory
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS TCP client application 0 on port 8001 ...

Now connected to IGate server euro.aprs2.net (185.43.207.219)
Check server status here http://185.43.207.219:14501

[ig] # aprsc 2.1.5-g8af3cdc
[ig] # logresp M0ZZM-10 verified, server T2HUN
[0L] M0ZZM-10>APDW15,WIDE1-1,WIDE2-1:!5118.12NS00003.02W#PHG2100Digipeater/iGate Testing
------------------------------------------

When I connect the IC-7100 to the Pi and list the USB ports, I obtain the following response:

:~ $ lsusb
Bus 001 Device 013: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Bus 001 Device 012: ID 08bb:2901 Texas Instruments PCM2901 Audio Codec
Bus 001 Device 011: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Bus 001 Device 010: ID 0451:2046 Texas Instruments, Inc. TUSB2046 Hub
Bus 001 Device 004: ID 062a:4c01 MosArt Semiconductor Corp. 
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
----------------------------------------------------

I tried to configure the PTT with options from COM6 to COM9 (the above screen paste shows COM10-COM14 because I turned the radio off and then on again, I suppose I need to reboot to reset the port numbers) but had no luck. I also tried COM1-COM4 just in case, but still get the same error message (with the variable of the port number corresponding to the entry in the configuration file).

Can anyone spot what I am doing wrong?

Thanks

Raul

wb2osz, 2019-08-16:
The "Issues" section is for tracking software defects and enhancement requests.  Questions like this should go to the discussion group.   https://groups.yahoo.com/neo/groups/direwolf_packet/info 

USB to serial adapters show up as /dev/ttyUSB0, /dev/ttyUSB1, and so on.   /dev/ttyS0, /dev/ttyS1 would show up for a desktop PC with built in serial ports, now increasingly rare.  

This doesn't look right:   Audio input device for receive: stdin (channel 0)
Take a look at your ADEVICE line and you will probably see "-" which needs to be removed.

m0zzm, 2019-08-16:
Thank you, this has partly resolved the problems. And thank you for the pointer to the Yahoo group - I wasn't aware it existed. I will carry out a few more tests and I will then post my next questions in Yahoo.

Raul

ggramaize, 2019-10-02:
Hi m0zzm,

PTT on the IC-7100 is supposed to be controlled though CI-V interface as far as I've tested (successfully).

My setup in 1.5 has hamlib support (standard debian package backported from bullseye to buster), and I've put the following in my configuration:

```
ADEVICE plughw:1,0
PTT RIG 370 /dev/ttyUSB0
```

It works like a charm so far.