wb2osz/direwolf #221: Timestamp in log file

geostant, 2019-09-01:
Hi,

I would appreciate if log file will include timestamps for TX/RX of packets.

Thanks

wb2osz, 2019-12-07:
The log file does have timestamps.    If you mean what is displayed on the screen, the "-T" command line option can be used to specify the desired format.  Does this help?

geostant, 2019-12-08:
The stdout that I see is pretty much like so:
```
Now connected to IGate server euro.aprs2.net (192.146.137.137)
Check server status here http://192.146.137.137:14501

[ig] # aprsc 2.1.4-g408ed49
[ig] # logresp 4X5DL-10 verified, server T2GB
[ig] 4X5DL-10>APDW15:!3225.73NR03453.13E&

4X5DL-7 audio level = 48(13/13)   [NONE]   _||||____
[0.2] 4X5DL-7>3R0P00,WIDE1-1,WIDE2-1:`>X<0x1c>l <0x1c>[/`_0<0x0d>
MIC-E, Human, Mic-Emsg, Special
N 32 00.0000, E 034 00.0000, 0 MPH
_0

```

So I'm missing timestamps for RX and TX here

radar27, 2020-03-05:
When you launch direwolf, append the switch -T %H:%M:%S.  This will add timestamps to in the HH:MM:SS format.  Note 14:56:22 below:

> VE1JOT-5 audio level = 50(24/22)   [NONE]   __:|:__
> [0.3 14:56:22] VE1JOT-5>APDW16,WIDE1-1:!4403.52NS06442.50W#PHG9140Brooklyn,N.S.
> Position, OVERLAY DIGI (green star) w/overlay S, 81 W height=20 4dBi omni
> N 44 03.5200, W 064 42.5000
> Brooklyn,N.S.
>

geostant, 2020-03-05:
Cool, thanks!

closed 2020-03-05 by geostant

pallas, 2020-08-10:
If you're looking for the full datetime, `-T '%FT%TZ'`.