wb2osz/direwolf #223: Cross compilation incompatibility

ew1abz, 2019-09-21:
@wb2osz, the master branch (version 1.5) is a little bit incompatible with a cross-compile approach. I made a successful attempt to build it (without telemetry toolkit, because it requires Perl) using Yocto build system. 
Some issues:
1. Hardcoded compiler
2. Hardcoded paths
3. Compile-time generation of fsk_fast_filter.h <- the hardest part!

If you are interested in, I'll prepare a pull request.

dranch, 2019-09-21:
Please try again using the Direwolf DEV branch which has significant changes in it.  From there, please create a Git merge request again DEV and your code will be reviewed.  Thanks!  If you have any Makefile related changes, please contribute to the Cmake effort being tracked here:

   https://github.com/wb2osz/direwolf/issues/216

--David

ew1abz, 2019-09-23:
David,
I've created a layer for Yocto/OpenEmbedded for Direwolf 1.5:
https://github.com/ew1abz/meta-direwolf 
It has fixes for cross-compilation (as patches). I guess it's good enough for now. Later I'll take a look at DEV branch and contribute there. 

For CMake is nice to have a start point. Moreover, it has to be for Linux, Windows, and Macosx. Maybe we can use GitHub actions for testing that.

ew1abz, 2019-10-13:
Yocto/OpenEmbedded layer for Direwolf 1.5 has been officially submitted:
http://layers.openembedded.org/layerindex/branch/master/layer/meta-direwolf/

closed 2019-10-13 by ew1abz