wb2osz/direwolf #224: Hamlib:  Error sending rig_set_ptt command for channel 0 PTT

craigerl, 2019-10-14:
libhamlib2 3.1-7
direwolf 1.4

Using hamlib RIG for PTT fails using icom7300.   Running rigctl for a moment while direwolf is running somehow fixes the problem (initializes the radio perhaps?).

`
craiger@pc:~$ direwolf
Dire Wolf version 1.4
Includes optional support for:  hamlib

Reading config file direwolf.conf
Audio device for both receive and transmit: plughw:6,0  (channel 0)
Channel 0: 300 baud, AFSK 1600 & 1800 Hz, D, 44100 sample rate / 3.
Ready to accept AGW client application 0 on port 8000 ...
Use -p command line option to enable KISS pseudo terminal.
Ready to accept KISS client application on port 8001 ...

Now connected to IGate server noam.aprs2.net (173.242.239.3)
Check server status here http://173.242.239.3:14501

[ig] # aprsc 2.1.5-g8af3cdc
[ig] # logresp KM6LYW verified, server T2NALA

Connected to KISS client application ...

Error sending rig_set_ptt command for channel 0 PTT
Communication timed out
[0L] KM6LYW>APDR14,GATE,WIDE2-1:=3854.68N/12056.13WL/A=001869 https://aprsdroid.org/
Transmit timing error: PTT is on 4026 mSec too long.
Error sending rig_set_ptt command for channel 0 PTT
Communication timed out
Error sending rig_set_ptt command for channel 0 PTT
Communication timed out
[0L] KM6LYW>APDR14,GATE,WIDE2-1:=3854.70N/12056.11WL/A=001797 https://aprsdroid.org/
Transmit timing error: PTT is on 4026 mSec too long.
Error sending rig_set_ptt command for channel 0 PTT
Communication timed out

.....
**run rigctld -m 373 -r /dev/ttyUSB0 -s 4800,            ctrl-c**
.....

[0L] KM6LYW>APDR14,GATE,WIDE2-1:=3854.70N/12056.11WL/A=001782 https://aprsdroid.org/

KM6LYW audio level = 18(9/10)   [NONE]
[0] KM6LYW>APDR14,GATE,WIDE2-1:=3854.70N/12056.11WL/A=001782 https://aprsdroid.org/
Position, PC user, APRSdroid Android App http://aprsdroid.org/
N 38 54.7000, W 120 56.1100, alt 1782 ft
 https://aprsdroid.org/

`

dranch, 2019-10-14:
Your Direwolf and Hamlib versions are quite old as Direwolf "master" branch is effectively v1.6 (it's quite stable) and HamLib is at v3.3 (even that is old as it's from August 2018 - the Hamlib file for the IC7300 was last modified on June 20, 2019 - https://github.com/Hamlib/Hamlib/blob/master/icom/ic7300.c ).  Anyway, I would highly recommend you compile to the newest versions of both versions and see if your problem still exists.  If it does, please report back.

craigerl, 2019-10-14:
Okay, built direwolf 1.6 (dev branch) against hamlib master, for completeness, no change.

```
openat(AT_FDCWD, "/usr/local/lib/libhamlib.so.4", O_RDONLY|O_CLOEXEC) = 3

craiger@pc:~$ direwolf
Dire Wolf DEVELOPMENT version 1.6 C (Oct 14 2019)
Includes optional support for:  hamlib cm108-ptt

Reading config file direwolf.conf
Audio device for both receive and transmit: plughw:6,0  (channel 0)
Channel 0: 300 baud, AFSK 1600 & 1800 Hz, D, 44100 sample rate / 3.
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS TCP client application 0 on port 8001 ...

Now connected to IGate server noam.aprs2.net (206.12.104.38)
Check server status here http://206.12.104.38:14501

Attached to KISS TCP client application 0...

Ready to accept KISS TCP client application 1 on port 8001 ...
[ig] # aprsc 2.1.4-g408ed49
[ig] # logresp KM6LYW verified, server T2BC
Error sending rig_set_ptt command for channel 0 PTT
Communication timed out
[0L] KM6LYW>APDR14,GATE,WIDE2-1:=3854.70N/12056.10WL/A=001726 craiger
Transmit timing error: PTT is on 4029 mSec too long.
```

craigerl, 2019-10-14:
Changing baud rate on radio back to default of 19200 fixes the problem.  Ideally direwolf could/should specify a bitrate in the config file?

ggramaize, 2019-10-15:
> Changing baud rate on radio back to default of 19200 fixes the problem. Ideally direwolf could/should specify a bitrate in the config file?

For the record, could you tell us what was the CI-V baud rate you were using with your problematic setup?

craigerl, 2019-10-15:
4800 baud, I have no good reason as to why, other than a flakey usb cable maybe.

It also explains why running rigctld momentarily (supplying 4800 baud rate) fixed the problem, as it likely set the port speed, then direwolf could use it.

ggramaize, 2019-10-15:
> 4800 baud, I have no good reason as to why, other than a flakey usb cable maybe.
> 
> It also explains why running rigctld momentarily (supplying 4800 baud rate) fixed the problem, as it likely set the port speed, then direwolf could use it.

Thanks, I'll make tests on my part to see if it's a model-specific issue (got an IC-7100 @home). It looks like hamlib expects the factory-default baud rate, then fails in a dirty way. Whichever the case, it seems to be more a hamlib issue, as far as I've inspected the source code.

swalberg, 2023-05-10:
Just found this which pointed things to baud rates being my problem, and was able to fix it without using rigctl by adding `stty -F /dev/ttyUSB0 speed 115200` to my script.