wb2osz/direwolf #225: dw-start on buster with rtl sdr won't start with cron job

geostant, 2019-10-18:
So I upgraded mu RPi to buster,
it's a Pi Zero W.

Been using it for igate for a while with RTL SDR without issues.

Running the following command on my terminal will work great:
`rtl_fm -f 144.39M - | direwolf -c sdr.conf -r 24000 -D 1 -`
but of course will catch my terminal and will terminate.

Opening a screen and running it inside works fine too,
I can detach and all is good.

However,
trying to set a cron job to run `dw-start.sh` or even running it directly without cron job yield no service - I can even see any error that can indicate what is the issue, the only output that I see on my screen when running it directly is this:
```
pi@pi-aprs-igate:~ $ ./dw-start.sh
Direwolf in CLI mode start up
There is a screen on:
	574.direwolf	(10/18/19 12:25:20)	(Detached)
1 Socket in /run/screen/S-pi.
-----------------------
```

running `screen -ls` will show that there are not detached screens
```
pi@pi-aprs-igate:~ $ screen -ls
No Sockets found in /run/screen/S-pi.
```

I also tried to `set -x` in the bash script to follow,
but still, there is nothing odd there that can indicate on an issue.

For now,
I just manually open a screen and run the service,
but that ain't production ready in my case.

dranch, 2019-11-05:
When things were running your previous version of Raspbian, how were you starting things?  If you read in the comments of the dw-start.sh script, it can be modified to auto-start the rtl_fm command.  You'll need to make those modifications yourself.  If you've already made that change to the script, what do you see in the /var/tmp/dw-start.log file?

rossmckelvie, 2020-07-05:
I had the same issue on Buster and it has to do with how the command is getting interpreted and passing thru stdin, trying to add this onto the end of a screen command. There was a workaround [posted in another issue](https://github.com/wb2osz/direwolf/issues/100#issuecomment-429674018), and although the direwolf process started correctly it was not inside a screen.

I was able to fix this by creating a file `start.sh`:
```shell
#!/bin/bash

rtl_fm -f 144.39M - | direwolf -c /home/pi/direwolf/sdr.conf -r 24000 -D 1 -
```

Then in `dw-start.sh`, I set `DWCMD="/home/pi/start.sh"`