wb2osz/direwolf #228: Support GeneralPlus USB sound cards

anm, 2019-11-16:
I was wanting to use the cmedia 108 PTT support so tried buying several usb sound dongles that I thought would be cmedia, but it seems that these devices now use a chip called GeneralPlus and I can't find a cmedia one. It would be good if these could be used for PTT too.

They seem to have USB idVendor=1b3f and idProduct=2008.

I have three different designs. All of them are COB (black blob on the PCB) so any extra GPIO lines cannot be accessed, if they even exist. They do have status LEDs though. I am hoping the LEDs can be controlled from the host and then these lines can be used for PTT.

dranch, 2019-11-16:
There are many many CM108-based audio devices available from many sellers.  If your bought a device that was supposed to be CM108 based and it turns out it was something else, that's fraud and I would return it to the seller (if you can).  

Anyway, there isn't a lot of information out there on these chips but it seems they are supported under Linux ( search within https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/cm108-type?view=all ).  To your other point, if your sound device uses the black blob type covering, your chances of connecting to any GPIO pins, etc. is VERY unlikely and also not worth it compared to getting a CM108 based device for probably the same cost.  Please also note that not all CM108-based sound devices perform the same with Direwolf both system-behavior wise and audio-quality wise.  Please read the UserGuide.pdf document for tested, recommended devices.

anm, 2019-11-16:
I've not been able to find a CM108. I've tried buying three different designs now from ebay and aliexpress. They all look the same as ones that used CM108, but apparently they don't now. I suspect that this GeneralPlus chip has appeared / become cheaper so the manufacturers have switched to it, and so if possible it would be worth supporting.

I wasn't going to try and expose the chip, but there are LEDs and in some cases switches. These lines could be used for PTT I hope, assuming they can be controlled by the host.

anm, 2019-11-16:
The devices I am talking about look like these:
https://www.aliexpress.com/item/32492176855.html
https://www.aliexpress.com/item/1823823296.html
https://www.aliexpress.com/item/33056020608.html

dranch, 2019-11-16:
Use your favorite Internet search engine and look up "SYBA SD-CM-UAUD".  These are known to work well with Direwolf.

anm, 2019-11-16:
I've tried patching cm108.c to recognise the VID/PID of this chip as a cm108, on the the chance they may be close enough to just work but no luck.

Running direwolf gives;
"Write to /dev/hidraw0 failed, n=-1, errno=32"

dranch, 2019-11-16:
Not sure why you would think that would work as these are completely different chipsets.  Each requires different driver code within the Linux usb-audio kernel module to work properly:

https://linux-hardware.org/index.php?id=usb:1b3f-2008

dafula, 2019-11-17:
> I've not been able to find a CM108. I've tried buying three different designs now from ebay and aliexpress. They all look the same as ones that used CM108, but apparently they don't now. I suspect [...]

These are with CM108. I have a couple of them and they work fine in packet and digi modes. The volume control is not linear - you can't have all 100 volume steps in alsamixer ( for example from 0 to 34 volume is muted)

https://www.ebay.com/itm/CM108-USB-Drive-Free-USB-Sound-Card-Laptop-Computer-External-Sound-Card-Module/201586324785

dranch, 2019-11-17:
GeneralPlus devices are supported by Linux but might not work with Direwolf (read the UserGuide.pdf file for more details).  Beyond that, if it can be proven that if GeneralPlus devices have GPIO pins and you point us to a data sheet of how to write a driver for it, we can keep this Git issue open.  If no, please *close* this GIt issue as Git is NOT used for supporting users.  Join and use the direwolf@groups.io email for all of your support needs.

wb2osz, 2019-11-24:
Mostly recapping what has already been said...  I've used USB audio adapters with the GeneralPlus chip and they work fine on Windows and Linux, no special drivers needed.  PTT is another matter.  They are just blobs not normal chips where you could tack solder on a wire to a GPIO pin.  Even if you had a model with extra LEDs or buttons that connect to the GPIO pins, we need some specifications on how software can access those GPIO pins.  I did some searching but was not able to find anything.  
You can reopen this if you find more information.

closed 2019-11-24 by wb2osz