wb2osz/direwolf #229: direwolf cannot use sound cards on OpenBSD

irl, 2019-11-24:
The cmake build allows direwolf to build successfully on OpenBSD, but it selects PortAudio as the backend and then tries to use a sndio device. This results in garbled sounds.

See http://www.sndio.org/ for the sndio API details.

(Everything works correctly if you feed the audio in from rtl_fm.)

chrisdebian, 2025-03-28:
@irl Do you know whether this is still true; if not, can the issue be closed?

wb2osz, 2025-03-29:
There are people using direwolf on OpenBSD.
Assuming this was fixed long ago.
Reopen still a problem.

closed 2025-03-29 by wb2osz

irl, 2025-04-02:
I have not looked at this for a while, if others are using it then this could be closed. It did require specific support for sndio to be implemented so that would also be a clue.