wb2osz/direwolf #233: Dev branch build: ptrdiff_t undefined

CIPop, 2019-12-01:
Likely caused by latest check-ins:

```
/home/pi/direwolf/src/dwgpsnmea.c: In function ‘dwgpsnmea_init’:
/home/pi/direwolf/src/dwgpsnmea.c:162:74: error: ‘ptrdiff_t’ undeclared (first use in this function)
    e = pthread_create (&read_gps_tid, NULL, read_gpsnmea_thread, (void*)(ptrdiff_t)s_gpsnmea_port_fd);
                                                                          ^~~~~~~~~
/home/pi/direwolf/src/dwgpsnmea.c:162:74: note: ‘ptrdiff_t’ is defined in header ‘<stddef.h>’; did you forget to ‘#include <stddef.h>’?
/home/pi/direwolf/src/dwgpsnmea.c:62:1:
+#include <stddef.h>
 #include "serial_port.h"
/home/pi/direwolf/src/dwgpsnmea.c:162:74:
    e = pthread_create (&read_gps_tid, NULL, read_gpsnmea_thread, (void*)(ptrdiff_t)s_gpsnmea_port_fd);
                                                                          ^~~~~~~~~
/home/pi/direwolf/src/dwgpsnmea.c:162:74: note: each undeclared identifier is reported only once for each function it appears in
compilation terminated due to -fmax-errors=1.
make[2]: *** [src/CMakeFiles/atest.dir/build.make:232: src/CMakeFiles/atest.dir/dwgpsnmea.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 14%] Building C object src/CMakeFiles/aclients.dir/aclients.c.o
[ 15%] Building C object src/CMakeFiles/decode_aprs.dir/dwgpsnmea.c.o
/home/pi/direwolf/src/aclients.c: In function ‘main’:
/home/pi/direwolf/src/aclients.c:297:76: error: ‘ptrdiff_t’ undeclared (first use in this function)
      e = pthread_create (&client_tid[j], NULL, client_thread_net, (void *)(ptrdiff_t)j);
                                                                            ^~~~~~~~~
/home/pi/direwolf/src/aclients.c:297:76: note: ‘ptrdiff_t’ is defined in header ‘<stddef.h>’; did you forget to ‘#include <stddef.h>’?
/home/pi/direwolf/src/aclients.c:86:1:
+#include <stddef.h>
 
/home/pi/direwolf/src/aclients.c:297:76:
      e = pthread_create (&client_tid[j], NULL, client_thread_net, (void *)(ptrdiff_t)j);
                                                                            ^~~~~~~~~
/home/pi/direwolf/src/aclients.c:297:76: note: each undeclared identifier is reported only once for each function it appears in
compilation terminated due to -fmax-errors=1.
make[2]: *** [src/CMakeFiles/aclients.dir/build.make:63: src/CMakeFiles/aclients.dir/aclients.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:610: src/CMakeFiles/aclients.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:573: src/CMakeFiles/atest.dir/all] Error 2
[ 15%] Building C object src/CMakeFiles/decode_aprs.dir/dwgps.c.o
[ 16%] Building C object src/CMakeFiles/decode_aprs.dir/dwgpsd.c.o
/home/pi/direwolf/src/dwgpsnmea.c: In function ‘dwgpsnmea_init’:
/home/pi/direwolf/src/dwgpsnmea.c:162:74: error: ‘ptrdiff_t’ undeclared (first use in this function)
    e = pthread_create (&read_gps_tid, NULL, read_gpsnmea_thread, (void*)(ptrdiff_t)s_gpsnmea_port_fd);
                                                                          ^~~~~~~~~
/home/pi/direwolf/src/dwgpsnmea.c:162:74: note: ‘ptrdiff_t’ is defined in header ‘<stddef.h>’; did you forget to ‘#include <stddef.h>’?
/home/pi/direwolf/src/dwgpsnmea.c:62:1:
+#include <stddef.h>
 #include "serial_port.h"
/home/pi/direwolf/src/dwgpsnmea.c:162:74:
    e = pthread_create (&read_gps_tid, NULL, read_gpsnmea_thread, (void*)(ptrdiff_t)s_gpsnmea_port_fd);
                                                                          ^~~~~~~~~
/home/pi/direwolf/src/dwgpsnmea.c:162:74: note: each undeclared identifier is reported only once for each function it appears in
compilation terminated due to -fmax-errors=1.
make[2]: *** [src/CMakeFiles/decode_aprs.dir/build.make:102: src/CMakeFiles/decode_aprs.dir/dwgpsnmea.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 16%] Building C object src/CMakeFiles/decode_aprs.dir/serial_port.c.o
make[1]: *** [CMakeFiles/Makefile2:647: src/CMakeFiles/decode_aprs.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

Workaround:
Adding the following to `direwolf.h` makes it go away:

```C
#include <stddef.h>
```

DwaineGarden, 2019-12-02:
Confirming I get it on a Intel machine also.   Fix resolves the compile error.

DwaineGarden, 2019-12-02:
I created a pull request to add #include <stddef.h> to direwolf.h.

mdomsch, 2019-12-06:
PR #236 resolves this similarly to #235 in that the proper header is included, now everywhere it's actually needed rather than the always-included direwolf.h.

wb2osz, 2019-12-07:
This has been fixed.

closed 2019-12-07 by wb2osz