wb2osz/direwolf #240: KISS Mode / Handling Delays

jcrix, 2020-01-06:
I could be going about this wrong. But I've built a BBS using direwolf as the sound modem.  Everything works well for the most part. But I have no way to tell went direwolf *actually* transmitted my packet. Which clearly causes timing issues.  Is there a better way to go about this?  It would be nice if there was a way to track when a packet actually transmits so I could probably build this event driven wise.  Like a mode we could put it in so that when a packet to be transmitted is submitted direwolf would responds OK when it actually transmits. Or error if something bad happens.

ab0tj, 2020-01-06:
This is probably more of a feature request than an issue, but I'll add a "me too" to this. This would be a useful thing to have with the way I use direwolf for an APRS tracker.

wb2osz, 2020-01-07:
Look in the Dire Wolf User Guide, section 9.4.6, KISS "set hardware" command.

There is a command to obtain the current number of frames sitting in the transmit queue.  Does this help?

jcrix, 2020-01-07:
I tested the command. Yes this should work for what I need.  Thanks!

closed 2020-01-19 by jcrix