wb2osz/direwolf #241: direwolf dev branch will not build with gpsd 3.20

n7nix, 2020-01-06:
Using this version of direwolf on 1/06/2020
   wget https://github.com/wb2osz/direwolf/archive/dev.zip
Direwolf will not build with: gpsd -V
gpsd: 3.20 (revision 3.20)

[ 29%] Building C object src/CMakeFiles/gen_packets.dir/gen_packets.c.o
/usr/local/src/direwolf-dev/src/dwgpsd.c:65:2: error: #error libgps API version might be incompatible.
 #error libgps API version might be incompatible.

[ 36%] Building C object src/CMakeFiles/atest.dir/dwgpsd.c.o
/usr/local/src/direwolf-dev/src/dwgpsd.c:65:2: error: #error libgps API version might be incompatible.
 #error libgps API version might be incompatible.

n7nix, 2020-01-06:
$ grep -i gpsd_api /usr/local/include/gps.h
#define GPSD_API_MAJOR_VERSION	9	/* bump on incompatible changes */
#define GPSD_API_MINOR_VERSION	0	/* bump on compatible changes */
 grep -A 3-i gpsd_api dwgpsd.c
grep: 3-i: invalid context length argument
pi@raspberrypi:/usr/local/src/direwolf-dev/src $ grep -A 3 -i gpsd_api dwgpsd.c
// Also compatible with API 7 & 8 with conditional compilation later.

#if GPSD_API_MAJOR_VERSION < 5 || GPSD_API_MAJOR_VERSION > 8
#error libgps API version might be incompatible.
#endif

wb2osz, 2020-01-07:
Until I get a chance to evaluate this properly, ...

What happens if you simply remove the "#error ..." line?   Maybe the API changes are only for functions not used here.

n7nix, 2020-01-07:
I changed the upper API_MAJOR_VERSION number in this line in dwgpsd.c
#if GPSD_API_MAJOR_VERSION < 5 || GPSD_API_MAJOR_VERSION > 9
and direwolf builds fine.
I don't use any direwolf gps functionality so I'm not a good test subject but I'm happy with the build.

mpannen1979, 2020-01-20:
I just built a version of direwolf from dev, and I get the same error as n7nix.  I bypassed the version check for GPSD in dwgpsd.c file.

However, when I run direwolf, it just repeats message 'Waiting for GPSD...' over and over.

cgps and gpsd DO work by themselves in this case.  The GPS is working and locking, etc.

mpannen1979, 2020-01-22:
I am using archlinuxarm.

I worked around this by installing previous version of gpsd like this:
pacman -U http://ca.us.mirror.archlinuxarm.org/armv7h/community/gpsd-3.19-3-armv7h.pkg.tar.xz

mdomsch, 2020-01-30:
Fedora rawhide (will be 32) includes gpsd 3.20 with it's API version 9, so at present this fails there.  If there's a direwolf 1.6 release imminent, I'd love to include that in Fedora as well.

mdomsch, 2020-02-03:
Fedora 32 reverted gpsd to v3.19 to avoid the APIv9 changes this late in the release cycle. That buys another 3-4 months before this has to be addressed for Fedora's timetables.

wb2osz, 2020-06-21:
Caught up with API version 9 and 10 changes.  
Tested with most recent version here:  https://gitlab.com/gpsd/gpsd.git  

commit 74cf91cc98a00265af8a5fed9c7d68757953de2f in dev branch.

Leaving this open for a while in case there is any additional feedback.

DakotaNelson, 2020-09-07:
Thanks for the patch, wb2osz! Has there been any movement toward getting it merged into main?

wb2osz, 2020-11-05:
Caught up with the API changes in the latest libgps source code.  This is in both dev branches and the 1.6 release.

closed 2020-11-05 by wb2osz