wb2osz/direwolf #243: Packet parsing error; Error sending KISS message to client application on pseudo terminal.

devnulling, 2020-01-12:
There seems to be an issue with some packets (or special characters), that causes this error/overflow below. 

This is running with current head of master of direwolf with the following params on Ubuntu 18. 

```
direwolf -t 0 -p -l logs/
```

```
Error sending KISS message to client application on pseudo terminal.  fd=9, len=69, write returned 61, errno = 2

pt write: No such file or directory
```

After this error is thrown, the terminal will output this after every packet:

```
KISS SEND - Discarding message because no one is listening.
This happens when you use the -p option and don't read from the pseudo terminal.
```

Below are two example packets that have caused this error:
```
VEGAS audio level = 49(16/22)   [NONE]   ||||_|___
[0.2] VEGAS>APNU19:!3557.90NS11530.08W#PHG7700/W2,NVn, WA6TLW,Las Vegas, NV A=008420
Position, OVERLAY DIGI (green star) w/overlay S, UIdigi, 49 W height=1280 0dBi omni
N 35 57.9000, W 115 30.0800
/W2,NVn, WA6TLW,Las Vegas, NV A=008420

Digipeater WIDE1 (probably VEGAS) audio level = 49(18/23)   [NONE]   |||||____
[0.2] WB7DRJ>S6QRTU,VEGAS,WIDE1*,WIDE2-1:`,[5l#">/'"<D}|%/%6'N|!wx>!|3
MIC-E, normal car (side view), Byonics TinyTrack3, In Service
N 36 12.4596, W 116 03.2532, 0 MPH, course 306, alt 2536 ft
Seq=378, A1=385, A2=591

Error sending KISS message to client application on pseudo terminal.  fd=9, len=69, write returned 61, errno = 2

pt write: No such file or directory

Digipeater VEGAS audio level = 50(16/23)   [NONE]   |||||____
[0.2] WB6QDI-9>APT311,VEGAS*,WIDE1-1,WIDE2-2:/231900z3609.23N/11509.71W>030/052/A=002073
Position with time, normal car (side view), Tiny Track III
N 36 09.2300, W 115 09.7100, 60 MPH, course 30, alt 2073 ft
KISS SEND - Discarding message because no one is listening.
This happens when you use the -p option and don't read from the pseudo terminal.

Digipeater VEGAS audio level = 50(16/23)   [NONE]   |||||____
[0.2] WB6QDI-9>APT311,VEGAS*,WIDE1-1,WIDE2-2:/231900z3609.51N/11509.61W>359/049/A=002076/TinyTrak3Plus
Position with time, normal car (side view), Tiny Track III
N 36 09.5100, W 115 09.6100, 56 MPH, course 359, alt 2076 ft
/TinyTrak3Plus
KISS SEND - Discarding message because no one is listening.
This happens when you use the -p option and don't read from the pseudo terminal.
```

```
Digipeater WIDE2 (probably VEGAS) audio level = 51(18/23)   [NONE]   ||||||___
[0.2] K7VQU-9>SVPVSX,N7ZEV-1,WIDE1,VEGAS,WIDE2*:`+ZEl O>/"9J}
MIC-E, normal car (side view), Unknown manufacturer, Off Duty
N 36 06.3800, W 115 02.4100, 0 MPH, course 51, alt 1660 ft

Error sending KISS message to client application on pseudo terminal.  fd=7, len=60, write returned 24, errno = 2

pt write: No such file or directory

Digipeater WIDE1 (probably N7ZEV-1) audio level = 59(22/18)   [NONE]   _|||||||_
[0.4] K7KSG>SV0SYY,N7ZEV-1,WIDE1*,WIDE2-2:`+*m!T`j/]"<4}=<0x0d>
MIC-E, JEEP, Kenwood TM-D710, En Route
N 36 03.9900, W 115 14.8100, 63 MPH, course 268, alt 2484 ft
KISS SEND - Discarding message because no one is listening.
This happens when you use the -p option and don't read from the pseudo terminal.
```

dranch, 2020-01-19:
Your issue here isn't really a bug (at the moment) so please create a new email thread on the Direwolf groups.io list and we can continue troubleshooting the issue there.  

Regardless, since you're using the "-p" option, I assume you're using a Linux based system.  What hardware, Linux distro and kernel version are you using?   In addition to that, please rerun your tests with the new Direwolf "develop" branch as it has major updates, improvements, etc.   If you still see the issue, once the system faults, can you confirm that your AX25 interface is still kissattach is still running (I doubt) it.

devnulling, 2020-01-19:
Setup:
- Baofeng HT
- SignaLink USB
- ubuntu 18.04
- kernel 5.0.0

I've updated to the dev branch and will follow up with more details if I run into it on this branch. 

Thanks!

wb2osz, 2020-01-20:
Nothing related to this has been changed for a very long time.
Why are you using the "-p" option?  Is it really needed?  What application is reading from the pseudo terminal?