wb2osz/direwolf #246: Dev Compile Errors - tocalls

jcrix, 2020-01-19:
When issuing cmake ..

I run into the following:

CMake Error at /usr/share/cmake-3.5/Modules/ExternalProject.cmake:1888 (message):
  Could not extract tarball filename from url:

    http://www.aprs.org/aprs11/tocalls.txt
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/ExternalProject.cmake:2459 (_ep_add_download_command)
  data/CMakeLists.txt:41 (ExternalProject_Add)

dranch, 2020-01-19:
Hmmm.. how old is your copy of the DEV git repo?  Run a "git pull" to make sure you have the newest code.  Beyond that, I'm not seeing this issue.  Direwolf only requires Cmake 3.1.x or newer so it should work.  Does the command "wget http://www.aprs.org/aprs11/tocalls.txt" work on this same machine?

--David

   make update-data

--David

jcrix, 2020-01-19:
Yes its current. I just pulled it. And yes I have newer cmake v 3.5.2. And yes the wget works. That was the first thing I tried.

dranch, 2020-01-19:
Can you share details about your system?  Hardware type, distro, distro version?   The "cmake .." and "make -j8" stages work fine here on Centos 6.10 (cmake 3.6.1) and Raspbian 10.2 (cmake 3.13.4).   Looking around, it might be an issue with your version of Cmake.  Any chance you can upgrade to something newer?

jcrix, 2020-01-19:
Yes thats the issue... the version of cmake.

closed 2020-01-19 by jcrix