wb2osz/direwolf #247: Please allow disabling of terminal colors

damaltor, 2020-01-25:
Hi,
i love direwolf but i really dont like the color scheme - for personal reasons of course, but for practical reasons too (power saving on amoled displays by leaving the background black). I would find it great if there was a configuration option to disable the white background and simply use the terminal colors the user configured for himself:
- consistent colors of all termional windows including direwolf
- respecting the users choice of color
- power saving possibility for amoled users
- maybe reducing the risk for issues like https://github.com/wb2osz/direwolf/issues/215
Keep up the nice work!
da_maltor

LongnoseRob, 2020-01-25:
have you tried using the option -t 0  ?

damaltor, 2020-01-25:
omg. Thanks a bunch. i promise to rtfm.

closed 2020-01-25 by damaltor

larsks, 2023-08-06:
I was surprised that this wasn't working; then I discovered that `-t 0` is not the same as `-t0`. That's unexpected!