wb2osz/direwolf #248: Height off by 10 on Beacon

wcravens, 2020-02-07:
I configured the following beacon...
PBEACON delay=1 every=30 overlay=S symbol="House QTH" lat=40^07.389N long=88^18.759W power=50 height=30 gain=8.3 comment="" via=WIDE2-1

But then it goes out on the air the height is reported as 40.  At first I thought it was just a glitch or that the packet got corrupted over the air.  But it isn't.  It reports height=40 every time.  It also reports power as 49.

dranch, 2020-02-07:
APRS PHG settings like height, power, etc. are not arbitrary.  There are limited, specific options allowed:

   http://aprsisce.wikidot.com/phg

I don't believe this is a bug and if you agree, please close this issue.

wcravens, 2020-02-07:
Roger, Roger.  I'll close the issue.  This could be good fodder for the documentation though.

closed 2020-02-07 by wcravens