wb2osz/direwolf #250: [Feature request] 16-QAM 6400bps modem

ggramaize, 2020-02-11:
Hi,

I'm currently planning field tests with the 4800bps modem, and while reviewing external documentation, I've seen 16-QAM signals seem have a similar noise to BER curve as 8-PSK waveforms. Thus, my proposal is to add a 6400bps modem (16-QAM @ 1600bauds over a 1800Hz carrier, as does the V.27 modem) to direwolf to compare the performance of those two modulations.

Kind regards

dranch, 2020-08-03:
This could be interesting and some QAM modems are available in the Linux Qsstv program for digital SSTV pictures.  This code was ported from TRXAMADRM so I think there is a history of code sharing, etc. but I have no idea how appropriate this code would be.

wb2osz, 2020-11-05:
It would be an interesting project and I've put some thought into it.  The part I can't figure out is how to properly phase lock a local oscillator for the demodulating.

ggramaize, 2020-11-05:
> The part I can't figure out is how to properly phase lock a local oscillator for the demodulating.

Is your issue related to coarse acquisition, or fine tracking? As far as I can remember what we did during our signal processing classes, we circumvented the coarse acquisition issue either using a QPSK gold code preamble (also interesting if you want to do dynamic rate detection) or using an adjacent unmodulated side tone, (also useful for channel estimation, especially if you want to propose this for lower rates on fading channels. IIRC, fldigi has such an mechanism -- as an option -- to optimize the phase tracking of their 8-PSK waveforms).

guitarpicva, 2022-04-28:
Some experimental multi-carrier codecs use a pilot "tone" to orient the detector.

wb2osz, 2025-03-30:
I don't understand how to do this.
Perhaps you could contribute some working code that could be adapted.

closed 2025-03-30 by wb2osz