wb2osz/direwolf #252: AX.25 version 2.0 segmenter

jmkristian, 2020-02-25:
Direwolf segments large frames when sending AX.25 version 2.0 packets. It should not. Instead, it should simply divide the data into packets without segment headers, on an AX.25 version 2.0 connection.

From https://www.tapr.org/pdf/AX25.2.2.pdf :
"The receipt of an XID response from the other station establishes that both stations are using AX.25 version 2.2 or higher and enables the use of the segmenter/reassembler and selective reject."
"The segmenter/reassembler procedure is only enabled if both stations on the link are using AX.25 version 2.2 or higher."

An example log.txt is attached. Direwolf is configured with PACLEN 64 and V20 W3XSC-1.
[log.txt](https://github.com/wb2osz/direwolf/files/4248795/log.txt)

jmkristian, 2020-02-25:
In the example above, Direwolf should have transmitted:

(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)..............10..............20..............30..............40
(I cmd, n(s)=1, n(r)=1, p=0, pid=0xf0)..............50..............60..............70.............7f<0x0d>

wb2osz, 2020-04-14:
Thanks for pointing that out.
Fixed in the 'dev' branch.
c15903edb9318e5d8cf580401fde03be788c3797

closed 2020-04-14 by wb2osz